openwrt/luci #4503: Wrong version numbers in packages for 19.07.4

ChristophFranzen, 2020-10-08:
Hello,

the version numbers of all Luci packages in 19.07.4 are "git-20.2**47**.75781-0d0ab01-1". I ran previously 19.07.2, where they were "git-20.2**72**.48698-9883bb3-1". Because the new "247" is a smaller number than the old "272", the packages get downgraded when re-installed. I got an inconsistent Luci installation this way with old and new version packages mixed. Obviously the number was meant to be: "git-20.2**74**.75781-0d0ab01-1". Fortunately it still runs when mixed up.

Regards, Christoph

hnyman, 2020-10-08:
Two things:

* It is automatic versioning based on date of last change in that app. Year 2020's 274th day ...
The version varies between different LuCI apps.
The date gets skewed by the initialisation of the source repo in each buildslave in buildbot. And that may cause inconsistent versions sometimes.

If you did have 272 in the old 19.07.2, you had recently opkg upgraded some apps (as native 19.07.2 is much older). But 19.07.4 was built from the head of the 19.07 branch and had the newest packages as of four weeks ago when it was built.

hnyman, 2020-10-08:
Ps. 
And a sysupgrade always upgrades the whole LuCI. No old code gets left there. So you have got a coherent version of LuCI.

jow-, 2020-10-08:
Buildbots continuously rebuild package repos, even for releases - therefor it is entirely possible that an image contains older packages compared to which are available in the repository.

Closing this as this is not something which we can fix in LuCI.

closed 2020-10-08 by jow-

ChristophFranzen, 2020-10-08:
Hello Jo-Philipp,

you probably didn't understand: I believe that the buildbot has corrupted the version number, not that it picked an older version!

It seems impossible that the buildbot used a package from day 247 on day 274 (or even later). Weh interpreting the following numbers (from git, I assume): 48698-9883bb3 and 75781-0d0ab01, you'll see that the lattzer is bigger, not smaller. This means that the buildbot or something else (what makes the ipk-packages?) has turned the 274 into a 247. This is the only interpretation that makes sense.

This is of course not a Luci bug, so please tell me where to report it, if you close it here.

jow-, 2020-10-08:
The day is derived from the date of the Git hash. 

```
$ git checkout 9883bb3 >/dev/null; ./build/mkrevision.sh 
HEAD is now at 9883bb397 treewide: i18n - backport and sync translations
git-20.272.48698-9883bb397
$ git checkout 0d0ab01 >/dev/null; ./build/mkrevision.sh 
Previous HEAD position was 9883bb397 treewide: i18n - backport and sync translations
HEAD is now at 0d0ab01a6 luci-app-opkg: flush menu cache after opkg actions
git-20.247.75781-0d0ab01a6
$ 
```
Seems to be as intended. I don't really understand your issue, even after reading it multiple times.

19.07.2 ships LuCI Git revision [13dd17fca](https://downloads.openwrt.org/releases/19.07.2/targets/x86/64/feeds.buildinfo) (git-20.057.55219-13dd17fca)
19.07.4 ships LuCI Git revision [0d0ab01a6](https://downloads.openwrt.org/releases/19.07.4/targets/x86/64/feeds.buildinfo) (git-20.247.75781-0d0ab01a6)

Version git-20.272.48698-9883bb3-1 is shipped by neither version, it comes from updated repository packages which track the latest branch head.