openwrt/luci #4506: luci-app-ddns: Time error

small-5, 2020-10-08:
The update time displayed by luci-app-ddns of the master branch is incorrect, which is inconsistent with the time displayed in the log file.
Version:git-20.280.43255-6483e80

feckert, 2020-10-08:
You have to give us more information

small-5, 2020-10-08:
The system time of my screenshot is 2020-10-08 20:13, but Last Update shows 2020-10-08 20:22.
[image]

feckert, 2020-10-08:
That means the time on your windows is different then the time on your router.
Have you set an ntp server?

small-5, 2020-10-08:
The router time is consistent with the windows time. The problem is not here, the time displayed by luci is not consistent with the updated time in the log file.

castillofrancodamian, 2020-11-01:
I can attest that the same thing happens here. The time is wrong.

Ansuel, 2020-11-01:
@feckert in theory the time from luci is taken directly from the script... If there is a bug it should be there. Did you found a way to reproduce this?

jow-, 2020-11-01:
Maybe a numeric overflow?

lorthirk, 2021-05-18:
I think that the timezone is calculated twice, somehow:

[image]

According to logs, last check happened at 09:25 and next one is due in 72 minutes, so at 10:37. I am GMT+1 (although with DST now, so it should be GMT+2).

closed 2021-05-19 by feckert