openwrt/luci #4522: PPPoE "keepalive"'s arguments look reversed

platform-717, 2020-10-17:
OpenWRT's [documentation](https://openwrt.org/docs/guide-user/network/wan/wan_interface_protocols#protocol_pppoe_ppp_over_ethernet) specifies:

> First number is for “seconds between each ping” (ppp's lcp-echo-interval option). Second number is “we assume the connection is down after this number of pings failed” (ppp's lcp-echo-failure option).

However, LuCI loads and saves these arguments in reversed order:

```
config interface 'wan'
        option proto 'pppoe'
        option keepalive '60 5'
        [...]
```

[image]

jow-, 2020-10-17:
The documentation is wrong in this case. LuCI's implementation matches the PPP protocol handler:
https://github.com/openwrt/openwrt/blob/master/package/network/services/ppp/files/ppp.sh#L128

Quick test:
```
root@OpenWrt:~# keepalive="60 5"; echo lcp_failure=${keepalive%%[, ]*}; echo lcp_interval=${keepalive##*[, ]}
lcp_failure=60
lcp_interval=5
root@OpenWrt:~# 
```

jow-, 2020-10-17:
I updated the wrong documentation in the wiki.

closed 2020-10-17 by jow-

platform-717, 2020-10-17:
Thanks for the clarification and update!

I was a bit confused because in my case setting *interval* = 5 and *failure* = 60 causes the line to randomly disconnect in a few minutes (while 5 is a common value for *interval* as per my knowledge), but setting them in reverse solved my issue.