openwrt/luci #4524: Luci chooses WPA3

castillofrancodamian, 2020-10-17:
When I try to connect to a network when I use the network browser even though it does not use WPA3, it always chooses it.
Use snapshots.

ptpt52, 2020-10-31:
yes, I notice this issue, maybe it is a bug.
ping @Ansuel @jow-

Ansuel, 2020-10-31:
@castillofrancodamian can you post a screenshot or a gif of the problem?

ptpt52, 2020-10-31:
Scan wifi and try to connect to somewhat ssid (and it is WPA2), but the default config is set to WPA3.

I have to change from WPA3 to WPA2 manually

castillofrancodamian, 2020-10-31:
> Scan wifi and try to connect to somewhat ssid (and it is WPA2), but the default config is set to WPA3.
> 
> I have to change from WPA3 to WPA2 manually

That's what I meant, and I can't get a screenshot.

Ansuel, 2020-10-31:
i would have preferred a screenshot but fin i will try to reproduce a push a fix for this.

castillofrancodamian, 2020-10-31:
> i would have preferred a screenshot but fine i will try to reproduce and push a fix for this.

It is easy to reproduce. Following the logic of connecting from the search engine, it can be observed.

Ansuel, 2020-10-31:
@castillofrancodamian can you tell me if your case with what is saved by default if you don't modify the encryption?
I reproduced the bug but it looks like the bug is not really the fact that it wrongly show wpa3 instead of the right one but the fact that it doesn't recognize the right encryption at all and save the first one (in my case luci set the connection with owe with a connection that should be wpa2)

castillofrancodamian, 2020-11-01:
@Ansuel At no point did I say that it does not show the correct encryption, but that it always chooses WPA3 even if the network to connect does not use that encryption and I have to manually change the corresponding one.

Ansuel, 2020-11-01:
You are saying the same thing in different words...
I'm saying that luci wrongly set the wrong encryption.
You are saying that luci by default show wpa3 and you need to change that manually.

If you don't change wpa3 manually luci gen a conf with sae set ahahha
Anyway i think i have fixed this and i will push a fix after some test

castillofrancodamian, 2020-11-01:
@Ansuel Maybe I have used the same words because I am using Google Translator. What I mean is that it does not show or choose the encryption in LuCI correctly.

Ansuel, 2020-11-01:
I mean that we are saying the same thing. Anyway found the problem will be fixed as soon as jow merge the fix I will push ;)
Thx a lot for the report

castillofrancodamian, 2020-11-01:
@Ansuel Thanks for giving me attention! 😊😊😊

PR #4561 (2020-11-01): Fix one display bug for ap join and #4524
Extensive description the commits log. I think i wrote too much there... ahahah
@jow- 

Fixes #4524
commit ba98a2fd (2020-11-01): luci-mod-network: fix logic bug in parse enc for network join

In handleJoinConfirm while is_wep is a boolean, is_psk and is_sae are an array. In the following if check, all 3 are used as boolean but Js treat empty array as positive values and this cause the ui to wrongly set the encryption to sae. fix this by checking if the array actually contains data.

Fixes: #4524
closed 2020-11-01 by jow-