openwrt/luci #4534: /usr/lib/lua/luci/dispatcher.lua:370: /etc/config/luci seems to be corrupt, unable to find section 'main'

swg0101, 2020-10-22:
Just getting this weird message on my latest master build today (this was compiled after a **make dirclean**). Haven't seen this issue on my last build two weeks ago so it sounds like this might be something new. This was seen on my build on the MobiPromo CM520 device.

`:~/openwrt/feeds/luci# git log -1
commit 2bbde8568a079b90d25f8155c48cf4eac7929fb9 (HEAD -> master, origin/master, origin/HEAD)
Author: Hannu Nyman <hannu.nyman@iki.fi>
Date:   Mon Oct 19 18:00:36 2020 +0300

    luci-app-dockerman: set fallback default of rollback to 90s

    Set the fallback value of the config change rollback timeout
    to 90 seconds to match the change in /etc/config/luci by commit
    81cf99a50.

    That commit changed the value in the config file, but did
    not change the underlying fallback values that do get applied
    when there is no proper config item in etc/config/luci.

    Users sysupgrading from old systems may have carried an ancient
    /etc/config/luci (without rollback config) with them, so this
    change should help them to see the intended user experience.

    Signed-off-by: Hannu Nyman <hannu.nyman@iki.fi>
`

Here are the outputs from the router directly:

`
root@OpenWrt:~# service rpcd status
running

root@OpenWrt:~# service uhttpd status
running

root@OpenWrt:~# uci show luci.main
luci.main=core
luci.main.lang='auto'
luci.main.mediaurlbase='/luci-static/bootstrap'
luci.main.resourcebase='/luci-static/resources'
luci.main.ubuspath='/ubus/'

root@OpenWrt:~# uci show rpcd
rpcd.@rpcd[0]=rpcd
rpcd.@rpcd[0].socket='/var/run/ubus.sock'
rpcd.@rpcd[0].timeout='30'
rpcd.@login[0]=login
rpcd.@login[0].username='root'
rpcd.@login[0].password='$p$root'
rpcd.@login[0].read='*'
rpcd.@login[0].write='*'

root@OpenWrt:~# free -m
              total        used        free      shared  buff/cache   available
Mem:            497          46         409           0          41         406
Swap:             0           0           0

root@OpenWrt:~# cat /etc/config/luci

config core 'main'
        option lang 'auto'
        option mediaurlbase '/luci-static/bootstrap'
        option resourcebase '/luci-static/resources'
        option ubuspath '/ubus/'

config extern 'flash_keep'
        option uci '/etc/config/'
        option dropbear '/etc/dropbear/'
        option openvpn '/etc/openvpn/'
        option passwd '/etc/passwd'
        option opkg '/etc/opkg.conf'
        option firewall '/etc/firewall.user'
[... 590 lines cut ...]
unbound-anchor - 1.11.0-4
unbound-checkconf - 1.11.0-4
unbound-control - 1.11.0-4
unbound-control-setup - 1.11.0-4
unbound-daemon - 1.11.0-4
unbound-host - 1.11.0-4
unzip - 6.0-8
urandom-seed - 2
urngd - 2020-01-21-c7f7b6b6-1
usbutils - 007-11
usign - 2020-05-23-f1f65026-1
wget - 1.20.3-4
whereis - 2.36-2
wireguard - 1.0.20200908-1
wireguard-tools - 1.0.20200827-2
wireless-regdb - 2020.04.29-1
wireless-tools - 29-6
wpad-basic-wolfssl - 2020-06-08-5a8b3662-12
xfrm - 2
zlib - 1.2.11-3
`

Does anyone have any ideas on what I should try?

Thank you!

jow-, 2020-10-22:
Like broken on the OpenWrt side, there's various ubusd regressions.

closed 2020-10-22 by jow-

kofec, 2020-10-22:
Probably related:
https://github.com/openwrt/openwrt/commit/2812ea3acb88c7e3649c912d6ad761bf8818fc51