openwrt/luci #4543: DUID length exceeds 36 chars when adding a static DHCP lease

dianlujitao, 2020-10-26:
My OpenWRT version is OpenWrt 19.07.4 r11208-ce6496d796

Here's a screenshot when adding a static DHCPv6 lease, obviously either the reported DUID from the pull-down menu or the internal length limit is wrong

[image]

hnyman, 2020-10-26:
Looks like 36 hex digits + colons, which apparently are displayed contrary to the validation rule.

This is from master, where the rule expects 20-36 hex chars without colons.  The length fails first, but the colons would cause failure too (as "hexstring" does not allow colons)

https://github.com/openwrt/luci/blob/a805a3178f902fe609d3d4a6c7b6b5b1dad88838/modules/luci-mod-network/htdocs/luci-static/resources/view/network/dhcp.js#L496

Ansuel, 2020-11-01:
@dianlujitao can we get more info? Are you using a special service to provide ipv6 dhcp?

dianlujitao, 2020-11-02:
> @dianlujitao can we get more info? Are you using a special service to provide ipv6 dhcp?

I use `dnsmasq-full` as DHCPv6 server.

dianlujitao, 2020-12-03:
bump, any updates?

Ansuel, 2020-12-03:
Do you still have this with latest master?

dianlujitao, 2020-12-04:
> Do you still have this with latest master?

Seems not with the latest 19.07.4, the colons are gone from the pull-down menu. But I reinstalled `odhcpd-ipv6only` because there are some minor issues when using `dnsmasq-full` alone, dunno if this happens to be a workaround.

hmoffatt, 2022-09-08:
This is still an issue. If you select a hex value from the drop-down list, they are not accepted as too long. Nor can you type in a full hex string value in the custom entry.

Also, the colons should not be stripped - dnsmasq needs the hex values with colons. Otherwise the values would be considered a string. The dnsmasq manual says

>  It is allowed to use client identifiers (called client DUID in IPv6-land) rather than hardware addresses to identify hosts by prefixing with 'id:'. Thus: --dhcp-host=id:01:02:03:04,..... refers to the host with client identifier 01:02:03:04. It is also allowed to specify the client ID as text, like this: --dhcp-host=id:clientidastext,..... 

I have tried stripping the colons and then the client IDs don't match. I put the hex string with colons in directly and then they match.

jow-, 2022-09-08:
Problem is that odhcpd expects DUIDs without colons afair, and that is the default DHCPv6 server on OpenWrt. This should be aligned on the OpenWrt side first before we can change it in LuCI.

hmoffatt, 2022-09-08:
@jow- true but even without that, there is a LuCI problem - it populates a list of DUIDs from the current leases, which may included colon-separated hex strings, but then doesn't accept the value you have just picked from the list.

jow-, 2022-09-08:
From which list of current leases? The dnsmasq ones? By default dnsmasq should not produce DHCPv6 leases because support for it is not compiled in, so it falls into the same category as above. Easiest fix would likely be making odhcpd accept and produce colon separated DUIDs to align with dnsmasq, but until this is the case we can't change it in LuCI to not break default setups.

At this point, dnsmasq based DHCPv6 should be simply considered unsupported.

hmoffatt, 2022-09-08:
I use dnsmasq-full, and I have the exact behaviour shown in the picture in the original report here (https://github.com/openwrt/luci/issues/4543#issue-729320773). The list is populated with full hex strings taken from active leases, but you can't select one because it is too long.

> At this point, dnsmasq based DHCPv6 should be simply considered unsupported in LuCI.

This particular issue could be fixed just by extending the length limit.

jow-, 2022-09-08:
Where are those colon strings come from exactly? I don't see this on my devices.

hmoffatt, 2022-09-08:
I see those values in /var/dhcp.leases. eg

```
1662681721 38:c9:86:40:86:37 192.168.42.18 potoroo ff:b7:b0:95:87:00:03:00:01:83:6d:06:a5:ef:01
```

hmoffatt, 2022-09-08:
> No, because DUIDs with colons are invalid for odhcpd.

Do you see values with colons suggested though?

jow-, 2022-09-08:
Nope, I do not see such suggestions

hmoffatt, 2022-09-08:
Is it harmful to allow longer values just so dnsmasq users? In each case users will be able to select from the list successfully.

gtxaspec, 2023-02-28:
my duid is 44 characters, example `0004000001580f0ceb16f323e54cbd318d66c7b33cbc` and odhcp recognizes it fine, and luci displays it, but complains it's too long.  Setting it manually in the dhcp config file works fine.  No colon's in my duid.  Generated from a t520 laptop running linux/intel amt.

closed 2024-07-26 by systemcrash

hmoffatt, 2024-07-27:
I have luci-mod-network `git-24.207.52553-1f36628` and I'm still seeing complaints about duids being too long. Is the fix not in this version yet?

systemcrash, 2024-07-27:
Try again tomorrow? You might need to build a new image. See whether that helps.

systemcrash, 2024-07-30:
ping @hmoffatt

hmoffatt, 2024-07-30:
It looks like this change is in `rpcd-mod-luci`, but I don't see an updated package for 23.05.4; I have `20240305-1`. Is there something else I can do to get this change?

systemcrash, 2024-08-05:
Available in 23 now.

hmoffatt, 2024-08-06:
opkg doesn't show it yet but I'll check again tomorrow.

PR #4642 (2024-07-26): rpcd-mod-luci: strip colon from dnsmasq duid
Dnsmasq DHCPv6 use colons in the generated duid present in the lease file. Strip the colon as this is not supported by the OpenWrt config and follow the same way odhcpd use to display duid.
@jow- 
Fixes: #4543
commit f71e9cbd (2024-07-26): rpcd-mod-luci: strip colon from dnsmasq duid

Dnsmasq DHCPv6 use colons in the generated duid present in the lease file. Strip the colon as this is not supported by the OpenWrt config and follow the same way odhcpd use to display duid.

Fixes: #4543
commit 64e19b2b (2024-08-05): rpcd-mod-luci: strip colon from dnsmasq duid

Dnsmasq DHCPv6 use colons in the generated duid present in the lease file. Strip the colon as this is not supported by the OpenWrt config and follow the same way odhcpd use to display duid.

Fixes: #4543

(cherry picked from commit f71e9cbd42db24c7743589804dacfdf5513ca883)