openwrt/luci #4551: Error - "No related RPC reply" (with ssl nginx luCI) - Raspberry Pi 4 snapshot

robcollins55, 2020-10-29:
Related to #3259 , the *No related RPC reply* error message in luCI with SSL-Nginx has returned to the luCI screen when logging into it with the browser.  I fresh-install my own build of the snapshot once every week (using the supported OpenWrt builder system) and this issue has popped up with my latest install with today's r14793-9f1927173a snapshot (all fine previous to this), so I thought I better report it.  Things I have tried so far:
- Restarting the browser (Firefox v82 64bit for Linux)
- Doing at ssh : `opkg update; opkg --autoremove remove luci-ssl-nginx; opkg install luci-ssl-nginx`
Please be specific and spoon feed me eg what file is of interest if you want an error log as I am a relative noob with helping with bugs :smile: .

Using the OpenWrt builder, my usual and current package lineup is as follows for building my image:
```console
$ make image PROFILE=rpi-4 PACKAGES="luci-ssl-nginx luci-app-acme acme ca-certificates php7-fastcgi uwsgi luci-theme-openwrt-2020 luci-app-statistics collectd-mod-thermal luci-app-sqm nano bash kmod-usb-net-asix kmod-usb-net-asix-ax88179 kmod-usb-net-cdc-ether kmod-usb-net-ipheth kmod-usb-net-kaweth kmod-usb-net-mcs7830 kmod-usb-net-pegasus kmod-usb-net-rtl8150 kmod-usb-net-rtl8152 kmod-usb-net-sr9700 luci-app-wireguard wireguard-tools wireguard qrencode luci-app-samba4 luci-app-ddns luci-app-adblock arp-scan arp-scan-database netdiscover kmod-fs-exfat kmod-fs-ext4 kmod-fs-ntfs kmod-fuse kmod-usb-storage kmod-usb2 kmod-usb2-pci kmod-usb3 kmod-fs-autofs4 block-mount blockd" FILES=files/
```
Any help/attention with this gratefully received :smile:

anomeome, 2020-10-30:
Should be [fixed](https://github.com/openwrt/packages/pull/13751)

robcollins55, 2020-10-30:
Thanks for prompt reply @anomeome that sounds awesome and thanks for the link..... I'll build another image next week based on the latest snapshot of that day and report back with yay/nay confirmation of my own experience.

robcollins55, 2020-12-01:
This issue indeed appears to have been resolved as I was able to install nginx LuCI with no issues.  Closing issue.

closed 2020-12-01 by robcollins55