openwrt/luci #4555: Luci errors on master

curtdept, 2020-10-31:
Fresh master branch build, Luci system rpc seems broken

[image]

hnyman, 2020-10-31:
I see the same with LuCI Master (git-20.304.73778-5a485f8) / OpenWrt SNAPSHOT r14809-37ef2197e6 

```
RPCError

RPC call to luci/getFeatures failed with error -32000: Object not found
  at handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-20.304.73778-5a485f8:11:3)
```

or

```
RPCError
RPC call to luci/getFeatures failed with error -32000: Object not found
at handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-20.304.73778-5a485f8:11:3)
at promise callback*parseCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-20.304.73778-5a485f8:9:70)
at promise callback*call (https://192.168.1.1/luci-static/resources/rpc.js?v=git-20.304.73778-5a485f8:5:110)
at declare/
```

And Firefox shows the exception/breakpoint as

> 
> Paused on exception
> RPCError: RPC call to luci/getFeatures failed with error -32000: Object not found at handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-20.304.73778-5a485f8:11:3) at promise callback*parseCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-20.304.73778-5a485f8:9:70) at promise callback*call (https://192.168.1.1/luci-static/resources/rpc.js?v=git-20.304.73778-5a485f8:5:110) at declare/</< (https://192.168.1.1/luci-static/resources/rpc.js?v=git-20.304.73778-5a485f8:23:320) at declare/< (https://192.168.1.1/luci-static/resources/rpc.js?v=git-20.304.73778-5a485f8:20:436) at probeSystemFeatures (https://192.168.1.1/luci-static/resources/luci.js?v=git-20.304.73778-5a485f8:188:162) at setupDOM (https://192.168.1.1/luci-static/resources/luci.js?v=git-20.304.73778-5a485f8:199:378) at promise callback*__init__ (https://192.168.1.1/luci-static/resources/luci.js?v=git-20.304.73778-5a485f8:152:362) at ClassConstructor (https://192.168.1.1/luci-static/resources/luci.js?v=git-20.304.73778-5a485f8:11:78) at (https://192.168.1.1/cgi-bin/luci:38:6)

ptpt52, 2020-10-31:
same problem.

peterwillcn, 2020-10-31:
same problem.
[image]

garypang13, 2020-10-31:
same problem.
[image]

MartB, 2020-10-31:
Can confirm this is indeed broken.

curtdept, 2020-10-31:
Rofl missing quotation mark, someone didn't run test.

jow-, 2020-10-31:
Fixed by #4560

closed 2020-10-31 by jow-

lynxthecat, 2021-09-25:
This is back, right(?):

RPC call to luci.wireguard/getWgInstances failed with error -32000: Object not found
  at ClassConstructor.handleCallReply (http://192.168.1.1/luci-static/resources/rpc.js?v=git-21.226.86205-376af36:11:3)

[image]

teqlee, 2021-09-26:
@lynxthecat 
Confirm issue come back
`RPCError
RPC call to luci.wireguard/getWgInstances failed with error -32000: Object not found
  at ClassConstructor.handleCallReply (https://192.168.16.1/luci-static/resources/rpc.js?v=git-21.267.65414-1d9067b:11:3)`

lynxthecat, 2021-09-26:
So this needs to be reopened, right? Let us know what extra info if any needed!

M95D, 2021-10-30:
I think this issue needs to be re-open.
I get the getFeatures error if busybox symlinks are installed in /bin instead of /usr/sbin. The error is caused by missing /usr/bin/env

rendiix, 2022-11-10:
same problem
[image]

lynxthecat, 2022-11-10:
Actually I have a slight suspicion this could be related to just needing to clean browser cache. Did you try that @rendiix?

A1EF, 2025-04-04:
I have same error on Xiaomi Redmi AX6S.
[image]

systemcrash, 2025-04-04:
I recommend clearing your caches or building a new image, because this problem is not endemic to the firmware or code. Probably a cache thing (since you upgraded packages, which can on occasion, break your luci setup). A possible fix is to uninstall luci and reinstall it.