openwrt/luci #4559: Can Not Upload ovpn File

mikeschumann, 2020-10-31:
When I try to configure an OpenVPN client, I click the browse button to select the opvn file.  I can successfully select the file, but the file name is not entered into the file name box.  If I try to manually enter the file name in the file name box, there isn't enough room for the full file name.

If I reduce the length of the file name, the browse button works correctly.

I am running OpenWrt 19.07.4 r11208-ce6496d796 / LuCI openwrt-19.07 branch git-20.247.75781-0d0ab01

dibdot, 2020-10-31:
Please elaborate the "file name". did you mean the instance name?

mikeschumann, 2020-10-31:
When you configure the OpenVPN client, you can use a .ovpn file from you windows computer.  You click on the Browse button to select the file from you hard drive.  If the path name of the file is too long, the pathname on the file is not saved in the box associated with the .ovpn upload function.  However, the file name does appear to the right of the Browse button.  If you click upload, you get an error that the file name box is empty.

Instead of using the browse button, you can manually enter the path name of the file in the file name box. However the box is WAY to short to let you enter anything but the simplest of path names.

dibdot, 2020-11-01:
The app uses the common file picker functionality, it's not intended for manual input. Just use a shorter path+filename if you've reached a limit here.

mikeschumann, 2020-11-01:
That is not a solution.  Using a shorter file name may be an awkward  workaround, but it is NOT a long term solution to the problem.  The current pathname length limit severely constrains the users ability to store ovpn files on their windows computers in a useful folder structure.

Mike Schumann 
mike@traditions.com
(651) 208-3791 (cell)

dibdot, 2020-11-01:
This is an open source project. If the functionality is not enough for you, try to improve it yourself and open a PR.

jow-, 2020-11-02:
I am not aware of any explicit file name restrictions we have in the upload handling, but maybe we do have some limited internal buffers in the parsing code or something...

Can you tell us the path / filename leading to the observed problem? I might try to replicate it on some Windows 10 VM.

danilvalov, 2022-12-08:
I had the same problem with uploading ovpn-file.
I found my problem: I used Chrome browser, and my OpenWRT admin panel had untrusted certificate (I just pressed `Open anyway` when Chrome asked me). And Chrome browser just skipped uploading on it side.

I opened the OpenWRT admin panel in Firefox, and my OVPN file was uploaded successful