openwrt/luci #4569: [luci-mod-dashboard] line offset in Wireless section

n8v8R, 2020-11-02:
Package: luci-mod-dashboard
Version: git-20.306.62653-4e86204

Package: luci-theme-openwrt-2020
Version: git-20.306.62653-4e86204
____

Tested with different browsers, all exhibiting the same

[image]

yppei, 2021-04-06:
I fixed it by commenting the position relative in the custom.css file.

.Dashboard .router-status-wifi .wifi-info .devices-info .tr .td:nth-child(3) {
 width:22%;
 overflow:hidden;
 /*position:relative;*/
top:-3px
}

stokito, 2023-06-10:
the ticket can be closed

adelton, 2025-01-11:
I assume this might have been addressed by 723eb386340c8da681aa2c5fb60101690b3231fc (for https://github.com/openwrt/luci/issues/6147).