openwrt/luci #4572: feature: add channel scan for wifi

Ansuel, 2020-11-04:
I notice that with the wifi scanning feature we have all the data to create a graph with that show the wifi channel utilization.
So i'm asking @jow- 
Do you think this feature would be useful? I would try to use the same code we use for the traffic graph so no external libs needed.

evelikov, 2020-11-05:
Fwiw I would greatly appreciate such a feature.

At the moment, one has to the wlan to monitor mode (via aircrack-ng) and use `iwlist scan` to get said data. Something which I had to do just a few moments ago :-)

Ansuel, 2020-11-05:
Actually this is already possibile using the WiFi join feature. I'm
searching a way to get channel width.

Ansuel, 2020-11-09:
Some WIP
[image]

Ansuel, 2020-11-16:
The ""Final"" Result.
A patch is needed for this to work. Now i just need to test this and polish the code. (improve readability since it's very compact 50-60 line)
[image]
[image]

jow-, 2020-11-16:
Very nice, great work!

Ansuel, 2020-11-16:
@jow- Where should i put this? Currently is integrated in the wifi scan... Is that right? Was thinking of moving it to the stats page

hnyman, 2020-11-16:
Statistics sounds wrong,
"scan" is better. 

Maybe create a "channel analysis" button next to "scan" in network/wireless?

Ansuel, 2020-11-16:
With statistics i mean the normal stats page, the one that show traffic in the interfacae...

Do you think a new button with Channel Analysis that show the graph + wifi (with info related to channel is more appropriate?)

hnyman, 2020-11-16:
>  With statistics i mean the normal stats page, the one that show traffic in the interfacae...

I got lost with that reference.
Which page you mean?

I see traffic count only on https://192.168.1.1/cgi-bin/luci/admin/network/network that does not have wireless.

Neither 
https://192.168.1.1/cgi-bin/luci/admin/status/overview
or
https://192.168.1.1/cgi-bin/luci/admin/network/wireless
has traffic counters for wifi.

I proposed 
https://192.168.1.1/cgi-bin/luci/admin/network/wireless

Or possibly it might additionally be a new page in main "Status" menu.

Ansuel, 2020-11-16:
> Or possibly it might additionally be a new page in main "Status" menu.

I was referring to the Realtime graph page in the status menu. A new page in the status menu is also a good idea.

hnyman, 2020-11-18:
I close this as the discussion can continue in PR #4598

PR #4598 (2021-01-07): luci-mod-status: add new channel analysis page
Add channel analysis support. This can be very useful as a user can directly use the webui to check wifi channel utilization without using external tool. This use data already provided by iwinfo.

@jow- This is the """final""" implementation. If you are fine with this (and after some feedback with some testing), we still need to decide where to put this.  @hnyman advised to put this in a separate status menu. Currently this is added to the join wifi modal.

(some details about how the page works, a tab with all the radios is created. The pool func [... cut]
commit 91a00084 (2021-01-07): luci-mod-status: add channel analysis support

Add channel analysis support. This can be very useful as a user can directly use the webui to check wifi channel utilization without using external tool. This use data already provided by iwinfo.

Fixes: #4572
closed 2020-11-18 by hnyman