openwrt/luci #4578: Existing dark themes and/or dark variant of existing ones

evelikov, 2020-11-06:
For a while I've been looking for a dark theme.

The only one I could find being darkmatter [1]. It is missing a proper package and might be having some odd issues. Sadly there's some friction [2] so it seems unlikely to have part of upstream soon.

Do you know any up-to date dark themes?  Would the luci team consider making a dark variant for any of the existing ones?

Thanks in advance

[1] https://github.com/apollo-ng/luci-theme-darkmatter
[2] https://github.com/apollo-ng/luci-theme-darkmatter/issues/18

feckert, 2020-11-06:
From my perspective I would like to have 2 themes.

- One with a menu tree at the top, like the luci-theme-openwrt.
- One with a menu tree on the side, like the luci-theme-openwrt-2020.

The other customization and properties changes, like colour and other CSS gimmicks could be controlled by css files.
As already considered in the [P/R](https://github.com/openwrt/luci/pull/4266). 

The problem with the diffrent themes is that once they are checked into the Luci feed, nobody cares about them.
Currently for example the theme luci-theme-matrial is not usable as I know by my last tests some time ago in the master does not support the ACL changed that were recently added into master.

evelikov, 2020-11-06:
I fully agree - dark variant/mode can be done completely via css "prefers-color-scheme". An attribute handled by all modern browsers.

jow-, 2020-11-06:
There is no dark theme atm

closed 2020-11-06 by jow-

teohhanhui, 2021-09-14:
I've created a UserCSS for `luci-theme-bootstrap` dark version: https://github.com/teohhanhui/openwrt-luci-bootstrap-dark

If the maintainers are okay with it, I'd like to send a PR. IMO this approach is low maintenance (just 100 lines of CSS, and they should be co-located with the corresponding style rules for the light theme), but with high visibility (enabled by default,so any issues would be reported by users).