openwrt/luci #4580: [luci-mod-status] CONFIG_NF_CONNTRACK_PROCFS being obsoleted in next LTS Kernel

n8v8R, 2020-11-06:
If understood correctly `admin/status/realtime/connections` leverages https://github.com/openwrt/luci/blob/5220d4f373ae70037bdf8f4e1934ec9631912e67/modules/luci-mod-status/src/luci-bwc.c#L431

however the next designated LTS Kernel 5.10 obsoletes `CONFIG_NF_CONNTRACK_PROCFS`

[image]

As a consequence there is no realtime graph for connections

[image]

feckert, 2020-11-06:
What are the options instead of getting this information from the system?

n8v8R, 2020-11-06:
Whilst it is obsoleting the kconfig is still present and may still work (did not compile/test with that option)

Else what `conntrack` provides http://conntrack-tools.netfilter.org/conntrack.html

e.g. events with `conntrack -E`

* `conntrack -L` > dumps the connection tracking table in /proc/net/ip_conntrack format 
* `conntrack -L -o extended` > dumps the connection tracking table in /proc/net/nf_conntrack format
* `conntrack -L -o xml` > dumps the connection tracking table in XML 

But that would require the _conntrack package_ as dependency

n8v8R, 2020-11-13:
Whilst the code is **considered** obsolete it has not been removed yet and still works with the kconfig enabled, thus perhaps nothing to be concerned about for now.

closed 2020-11-13 by n8v8R

fengyang0317, 2023-10-28:
Is is possible swith to `conntrack` now? I am trying to use lxc container to run openwrt. `/proc/net/nf_conntrack` is not available.