openwrt/luci #4582: LUCI DHCP Entries with multiple MAC not using new format

fzinken, 2020-11-06:
**Bug Description**
When entering a DHCP Static Lease with multiple MAC addresses LUCI adds it to `/etc/config/dhcp` in the old format.

**Example config** 
[image]

**LUCI generated**
```
config host
	option mac 'AA:AA:AA:AA:AA:AA BB:BB:BB:BB:BB:BB'
	option name 'Test'
	option dns '1'
	option ip '192.168.100.100'
```

**New Format**
```
config host
	list mac 'AA:AA:AA:AA:AA:AA'
        list mac 'BB:BB:BB:BB:BB:BB'
	option name 'Test'
	option dns '1'
	option ip '192.168.100.100'
```

LUCI also only show it correct if the new format is being used.

**LUCI display after saving** 
[image]

**LUCI display after manual correction**
[image]

Ansuel, 2020-11-25:
Thx for the report, to be more precise, the bug is in the fact that the custom form permit to insert a list of mac split with a space. Luci detect only the list form. Should be an easy fix.

Ansuel, 2020-12-06:
@fzinken had the time to test this and to my surprise I can't reproduce this. With your config the ui can correctly show the 2 mac. 
Can you confirm this?

fzinken, 2020-12-06:
> @fzinken had the time to test this and to my surprise I can't reproduce this. With your config the ui can correctly show the 2 mac.
> Can you confirm this?

I am confused, you  mean you can generate a double MAC entry with LUCI and it is correctly shown?
You mean this config `LUCI generated` is working for you?

Ansuel, 2020-12-06:
@fzinken Yes. Are you on master or on stable?

fzinken, 2020-12-06:
> Are you on master or on stable?

Assume stable.
luci - git-20.247.75781-0d0ab01-1

hnyman, 2020-12-06:
> Assume stable.
> luci - git-20.247.75781-0d0ab01-1

Yep, 0d0ab01 is from stable 19.07 branch

SampsonF, 2021-03-10:
I have the same issue.

OpenWrt 19.07.7 r11306-c4a6851c72 / LuCI openwrt-19.07 branch git-21.062.76689-a607f9c

tommie, 2021-11-27:
IIUC

* The [`form.Value`](https://github.com/openwrt/luci/blob/0b49ed45c6e9f4bc10abdcea392125aec3794e58/modules/luci-mod-network/htdocs/luci-static/resources/view/network/dhcp.js#L539) should be replaced by a `form.DynamicList` ([example in `wireless`](https://github.com/openwrt/luci/blob/0b49ed45c6e9f4bc10abdcea392125aec3794e58/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js#L1091)).
* [This line](https://github.com/openwrt/luci/blob/0b49ed45c6e9f4bc10abdcea392125aec3794e58/modules/luci-mod-network/htdocs/luci-static/resources/view/network/dhcp.js#L553) is what causes a list to be rendered as a space-separated list now, but the datatype [`list(macaddr)`](https://github.com/openwrt/luci/blob/0b49ed45c6e9f4bc10abdcea392125aec3794e58/modules/luci-mod-network/htdocs/luci-static/resources/view/network/dhcp.js#L540) signifies a space-separated value, as opposed to a UCI list.
* Would simply changing to `DynamicList` be able to parse old configs, or is there a need for migration logic? I suspect [`L.toArray`](https://github.com/openwrt/luci/blob/0b49ed45c6e9f4bc10abdcea392125aec3794e58/modules/luci-base/htdocs/luci-static/resources/luci.js#L3003) is enough to handle both cases.
* History: @Ansuel added the `toArray` in https://github.com/openwrt/luci/commit/6c9a6c334e2fa55f4b64b64b85e0b104f0b99037, but it didn't affect the write-side.
* Is there anything else that needs changing?

systemcrash, 2024-07-05:
Good catch! Should be fixed now, I think.

commit cef99564 (2024-07-05): luci-mod-status: Set status page "Set Static" for MAC to list type

MACs are list type in DHCP; 1 IP can have multiple MACs

See https://github.com/openwrt/luci/issues/4582#issuecomment-2209390439

Closes #4582
closed 2024-07-05 by systemcrash