openwrt/luci #4601: luci-proto-openconnect access denied

lordrak007, 2020-11-18:
Hello,
i have just installed openwrt, install luci-proto-openconnect package and add new interface of Open Connect.
Interface was added but not configuration window was open. When i click Edit button i have got error:
RPCError
RPC call to luci.openconnect/getCertificates failed with error -32002: Access denied
  at ClassConstructor.handleCallReply (https://192.168.0.1/luci-static/resources/rpc.js?v=git-20.319.48994-50b7ab5:12:3)

I am using Powered by LuCI openwrt-19.07 branch (git-20.319.48994-50b7ab5) / OpenWrt 19.07.4 r11208-ce6496d796
Than you

jow-, 2020-11-18:
Try logging out an in again, see if the error persists then.

feckert, 2020-12-02:
@lordrak007 have you tried the suggestion?
If this does not work please reopen

closed 2020-12-02 by feckert