openwrt/luci #4603: luci-app-mwan3: Duplicate metric warning on separate family interfaces and other potential false warnings

jamesmacwhite, 2020-11-18:
@feckert @aaronjg 

I've kind of ignored most of the warnings from luci-app-mwan3 as in my opinion some of them might be a bit false positive because it doesn't really handle IPv6 interfaces well from what I can see.

[image]

I have a couple of questions though.

1. Is having the same metric value applied to both the WAN and WAN6 interface really invalid? Because one is IPv4 and the other IPv6 surely this is fine, because it's two different routing tables for IPv4 and IPv6? I'm assuming it's perhaps just looking at the interface entry itself and thinks it's duplicated.

2. Any IPv6 interface seems to throw the `Interface xxxx has no default route in the main routing table`, again I think this because it's not specifically looking at the IPv6 configuration here.

All interfaces work fine, I have just been ignoring the errors but I've seen other users in the forums get put off by the above errors from LuCI, when I don't think they are that useful because I don't even think from an IPv6 perspective they are properly reporting an issue.

I'm not an expert however, so I wanted to ask for further clarification. Similar points around the diagnostics functionality, for any IPv6 interface most of the diagnostic options don't work. I personally don't use it that much and would always use SSH and test with various tools like traceroute, iperf3, ping etc.

Thank you!

jamesmacwhite, 2020-11-18:
I'm assuming my first point it's because the condition is looking at an IPv4 route of 0.0.0.0 present on the interface which won't be valid for an IPv6 interface.

https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/model/cbi/mwan/interface.lua#L85-L94

jamesmacwhite, 2020-11-18:
And the duplicate metric warning looks to be just looping through interfaces and seeing if the assigned metric value is present, not taking into account the interface being a different family. In fact the interface config itself doesn't appear to actually expose the interface family set in this case.

https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/model/cbi/mwan/interface.lua#L73-L83

aaronjg, 2020-11-20:
That all sounds correct - the diagnostics appear to be quite off for IPv6, and a P/R fixing this would be welcomed.

It might also be nice to add a diagnostic for IPv6 to make sure that the user has some sort of rewrite rule in place for their IPv6 addresses, since we are currently leaving it up to the user to pick the solution that works best for them, but we might want to flag it if they have not set anything up to do the IPv6 address rewriting.

jamesmacwhite, 2020-11-20:
My experience with LuCI development is limited, but I think the initial step would be to have the defined family option that is set on within an interface config available to the interface model as a first step, that is then going to allow for being able to target IPv4 and IPv6 conditions within the existing metric/default route conditions, as right now they are pretty much written from an IPv4 only perspective.

In addition, the controller outputting the IPv6 counter parts for things like the ip route commands.

https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/controller/mwan3.lua#L194

For IPv6 the only diagnostic option that works is pinging the tracking IP.

feckert, 2020-11-20:
As @aaronjg  has already said and that you @jamesmacwhite have already recognized this correctly.
This is still from the time when we only had ipv4 tracking or could configure this over the LuCI. 
Either we do the whole check out or we have to check if we use ipv4 or ipv6 tracking for this interface and depending on this show the warning on miss configuration.

jamesmacwhite, 2020-11-20:
@feckert I think at least initially the warnings need to be not shown for IPv6 related interfaces because they are a bit misleading and aren't testing anything IPv6 really. When users are trying to diagnose something with their mwan3 configuration and they those messages appear for IPv6 interfaces, it can lead you think something is wrong, when in fact this is probably not the case.

If the interface config can at least have some form of IPv6 awareness as a start that would be a good place to build on.

I think longer term you'd want to have IPv4 and IPv6 compatibility for where there are currently IPv4 only handling, mainly around the diagnostics and information shown like ip route and such, but there probably needs to be a way to determine IPv6 is both enabled generally and if IPv6 configuration in mwan3 is present before doing that.