openwrt/luci #4609: modal dialogs should be closeable via ESC key.

nerdoc, 2020-11-20:
All dialogs in luci, at least the ones which are modal, should be closeable via ESC, for a better user experience.
[image]

Especially when you are upgrading packages via luci-opkg, it's tedious to always find the (only) "Dismiss" button, which is always on another point on the y axis... ;-)

This is no bug, but an feature request/enhancement, which could be done very easily (depending on the codebase of dialogs) - I don't know the internals, am just a user.

And if so, maybe even use something like Enter for the default button (Apply, Ok etc)? This is a bit more complicated, as some elements could use that key as well.

jow-, 2020-11-20:
This cannot be easily implemented in a generic manner since it is not clear which action `esc` should perform in case there's multiple actions available.

Moving this to to utopia in case osmeone is willing to work on it.

tapper82, 2021-02-10:
https://edenspiekermann.github.io/a11y-dialog/example/
https://www.w3.org/TR/wai-aria-practices/#dialog_modal

tapper82, 2021-02-10:
Being able to close the modal with the esk key would be better.

systemcrash, 2021-12-08:
This one bugs the shit out of me.

systemcrash, 2021-12-08:
If Dismiss is the only button of a reasonably in-focus form, how about mashing ENTER or whichever key to accept?

tapper82, 2021-12-08:
This is a bug, an A11y one.

commit c9939a83 (2023-02-05): luci-base: ui.js: modal dialog tweaks

 - Automatically focus modal dialog frame
 - Close modal dialog on pressing escape key

Fixes: #4609, #6205
closed 2023-02-05 by jow-