openwrt/luci #4610: Network and RPC errors after snapshot install

timkgh, 2020-11-20:
I run snapshot builds on a UniFi AP. Every time I update OpenWRT and re-install luci and luci-ssl I have a hard time logging in to LuCI again, I keep getting various network or RPC errors, e.g.

```
NetworkError
HTTP error 400 while loading class file "/luci-static/resources/view/status/index.js?v=git-20.311.85590-5c5b134"
  at compileClass (https://unifiap/luci-static/resources/luci.js?v=git-20.311.85590-5c5b134:169:16)
```

After a few page reloads it will eventually start working.

I also noticed that after I manage to log in, the menu at the top of the page is missing. I have to refresh the page one more time to get it all back.

Browser console errors:

```
luci.js?v=git-20.311.85590-5c5b134:66 POST https://unifiap/ubus/?1605713311367 404 (Not Found)
(anonymous)	@	luci.js?v=git-20.311.85590-5c5b134:66
request	@	luci.js?v=git-20.311.85590-5c5b134:47
post	@	luci.js?v=git-20.311.85590-5c5b134:71
probeRPCBaseURL	@	luci.js?v=git-20.311.85590-5c5b134:186
__init__	@	luci.js?v=git-20.311.85590-5c5b134:152
LuCIClass	@	luci.js?v=git-20.311.85590-5c5b134:11
(anonymous)	@	(index):38
```

More details and config in this thread:
https://forum.openwrt.org/t/luci-network-and-rpc-errors-after-snapshot-install/79663

jow-, 2020-11-20:
Which browser do you use?

timkgh, 2020-11-21:
> Which browser do you use?

Chrome 87

> Update: investigated myself, seems to be specific to ustream-wolfssl.

Which `luci-ssl-*` flavor is recommended? `luci-ssl-openssl` maybe?
I just installed `luci-ssl` per openwrt wiki instructions.

timkgh, 2020-11-21:
Unfortunately I'm not able to install `luci-ssl-openssl` on snapshot builds:
```
 * check_data_file_clashes: Package libustream-openssl20200215 wants to install file /lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20200215
 * opkg_install_cmd: Cannot install package luci-ssl-openssl.
```

hnyman, 2020-11-21:
Uninstall "libustream-wolfssl" first, so that you can install luci-ssl-openssl instead (as it installs "libustream-openssl")

timkgh, 2020-11-21:
It's tricky, have to download libustream-openssl first otherwise if libustream-wolfssl is uninstalled one cannot download packages anymore. Even if I do that, I cannot install luci-ssl-openssl the UniFi AP seems to run out of space.

I appears the plan is to not use wolfssl going forward due to various issues with it. I will live with the errors for now until snapshots move away from wolfssl.

timkgh, 2020-11-21:
I think I found the problem, the default uhttpd config file has this option commented out:
`# option ubus_prefix '/ubus'`
If I uncomment it and restart uhttpd, I don't see any errors anymore.

jow-, 2021-12-03:
I suppose this is fixed now.

closed 2021-12-03 by jow-