openwrt/luci #4611: Associate client/station MAC to name when running in AP mode

timkgh, 2020-11-21:
This is a feature request for LuCI to provide a way to associate a friendly name to a MAC address for wifi clients when running in AP mode. The AP itself does not have an IP address on the normal VLANs (it only has a management IP but that is not useful for arp) and does not run dnsmasq either. The router on the network knows the names, IPs and MACs.

It would be useful to keep an association list MAC -> friendly name, in particular on a home network where all clients are pretty much static, so that it's easy to identify clients in the GUI.

jow-, 2020-11-21:
You can either populate /etc/ethers via SSH or declare static DHCP leases without IP (even if dnsmasq is unused) to associate names with MACs.

closed 2020-11-21 by jow-

timkgh, 2020-11-22:
Great, thank you. I went with `/etc/ethers` and it works well enough.