openwrt/luci #4615: luci-app-adblock: no bug, just suggestion:   present host names in "Domain" column of DNS Report as link anchors rather than as text for "OK" host names.

Bushmills, 2020-11-22:
Albeit host names aren't URLs, and may therefore not represent the target of the original request, just as resolved (or not) host names don't need to origin from a request for www targets, being quickly able to inspect the nature of a site by simply clicking on it may help deciding whether a host name from that list should deserve blacklisting.
What it takes for this is displaying the DNS report "Domain" column output as link anchors for those hostnames which resolved (showing as "OK") rather than as plain text. Presumably this should be an option a user should be a able to turn off if undesired (or rather, to turn on in other cases). I don't see compatibility issues, especially if such a feature needs to be turned on, and it appears to be a little effort addition.

dibdot, 2020-11-22:
In one of my adblock dev versions I've already implemented a similar approach: All domains (e.g. imap.1und1.de) in the report are search engine links, e.g. `https://duckduckgo.com/?q=imap.1und1.de`. This works for all domains listed and you prevent direct calls  to unknown external content.

Bushmills, 2020-11-22:
Substitution of a place holder variable in a configurable URL string could provide the flexibility a user may desire. Specifying only the place holder variable would then open the web server behind the host name, putting it in a query string allows to consult any search engine allowing arguments in urls, as well as allowing uncritical defaults.

dibdot, 2020-11-23:
I won't support direct calls - besides security concerns, many of them will simply not work and create countless requests.

dibdot, 2020-11-27:
Search anchor added with #4623

closed 2020-11-27 by dibdot

Bushmills, 2020-11-27:
Thanks, Dirk! Good action, I believe.