openwrt/luci #4624: Cannot save changes to time because of blank Design field

socketbox, 2020-11-28:
1. Logged in to router because devices that should have been on the network were reporting that no network was available.
  2. Saw that the time, in LuCi interface, was off by a month.
  3. Corrected time and went to save.
  4. Cannot save because the "Design" field in the "Language and Style" tab is empty.
  5. Attempt to enter _any_ value into said field. No joy.
  6. Reboot router like hopeful idiot.
  7. Try again. Fail.
  8. Look for information on this aspect of the web UI online. Hard to disambiguate results having to do with "design" at a lower level.
  
Firmware version: OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.242.67031-c3a0c60

[image]

hnyman, 2020-11-28:
That field is supposed to contain the selection of available themes..
There should be a section in /etc/config/luci like this:
```
config internal 'themes'
        option Bootstrap '/luci-static/bootstrap'
```
Bootstrap is the default theme, but you might also have others installed.

[image]

jow-, 2020-11-28:
This is caused by a corrupt /etc/config/luci. Does not happen with a stock image or after installing any theme.

closed 2020-11-28 by jow-