openwrt/luci #4627: luci-theme-openwrt-2020!

olimpiumarius, 2020-11-29:
Hello all,
This tab is not working in this 
[image]
[image]

hnyman, 2020-11-29:
You are apparently talking about a specific app, banip

cc author @dibdot

hnyman, 2020-11-29:
And you are apparently talking about the private derivative fork "swrt", instead of the main OpenWrt.

dibdot, 2020-11-29:
Well, I "think" this is an CSS issue of this theme with legacy apps. Anyway, currently I'm working on a JS version of this app, therefore I do not plan to do any changes on this old legacy code.

olimpiumarius, 2021-01-02:
This issue persists only on the new OpenWrt theme, too bad no one wants to solve this problem!
[image]
[image]
[image]
[image]
[image]
[image]
[image]

dibdot, 2021-02-04:
will be fixed with #4762

closed 2021-02-04 by dibdot