openwrt/luci #4632: Renaming a wireless interface with a period caused it to become disabled, and can not be enabled.

stuffa, 2020-12-01:
In wireless, interface, advanced settings: 
Renaming a wireless interface with a period caused it to become disabled, and can not be enabled.

Interface names with a period are usually interpreted as a VLANs.  But there are no VLANs in wifi interfaces.
If an interface is configured with a period (eg: wlan-2.4g) then the interface will fail to start.
renaming it to wlan-24g is ok.

The easiest solution is a validation on the LuCI page to stop the use of periods in wireless interface names.
It may not be possible to fix the underlying script as linux utilities used for wireless, may also be used for other interfaces that do support VLANs and will interpret the '.' as a vlan separator.   So it is wise to avoid the use of a period in wireless interface names .

feckert, 2020-12-02:
Can you please send me a screenshot where I can see where you type this. So that I can find the entry

jow-, 2020-12-02:
Not sure if it is worth fixing that corner case. First of all, using a dot is only one of many different ways you can create a breaking config. Adding colons, spaces or other chars will likely result in breakage as well.

Furthermore you could specify a name which is already taken, such as `eth0` which will also result in undefined behavior.

Since that ifname override option is placed in the `advanced` tab, users are expected to know what they're doing when setting it.

stuffa, 2020-12-03:
If a validator is added, it can easily be extended by others with less
familiarity with LuCI to add other cases over time, significantly enhancing
the service.

I will add one in the future myself, but I first have to become familiar
with the new LuCI JavaScript framework.  I'm still locked in the LUA
version.

Cheers

----------------------------------------------------------
Chris Martin
m: +61 419 812 371
e: chris@martin.cc
----------------------------------------------------------