openwrt/luci #4691: Luci-Theme-Rosy does not show in the menuconfig

Masterchiefm, 2020-12-27:
I use the master branch of openwrt, I see the luci feed contain a theme named rosy. But I cannot see it in menuconfig-> Luci -> Theme.

If Luci 20 do not support rosy, please remove rosy from luci themes.

nickberry17, 2020-12-27:
It is already marked `@BROKEN`, and does not show up in master for me.  Do you have `Show broken packages / platforms / devices` enabled in` advanced configuration options` of `menuconfig`?

hnyman, 2020-12-29:
It has been marked BROKEN already in 2019, so it does not get compiled in buildbot or be offered in menuconfig (unless you specifically set yourself to see also broken packages). With default settings you can only see it as source code among other themes. So it not showing in menuconfig is expected.

So far the rosy theme code has been there, BROKEN since Dec 2019, waiting for somebody to fix it, but so far nobody has stepped up. Jow pondered already in March that it might get removed soon, so maybe the time has come for that.

closed 2020-12-29 by hnyman