openwrt/luci #4697: luci-app-transmission: unable to set user and group

gw826943555, 2020-12-29:
[image]
I think it is caused by insufficient authority.

ak1bou, 2021-01-07:
It's easy to solve this problem:
```
sed -i '/"file": {/a\\t\t\t\t"/etc/passwd": [ "read" ],\n\t\t\t\t"/etc/group": [ "read" ],' /usr/share/rpcd/acl.d/luci-app-transmission.json
service rpcd restart
```
Then refresh the page.

idarek, 2021-08-30:
Hello,
I am experiencing same issue now. 
sed command above doesn't seams to work

> It's easy to solve this problem:
> 
> ```
> sed -i '/"file": {/a\\t\t\t\t"/etc/passwd": [ "read" ],\n\t\t\t\t"/etc/group": [ "read" ],' /usr/share/rpcd/acl.d/luci-app-transmission.json
> service rpcd restart
> ```
> 
> Then refresh the page.

This issue is shown on OpenWrt 21.02 and the solution above sort it. Thanks.

thekiefs, 2021-09-24:
Can we apply this patch please?

commit 1d9067b2 (2021-09-24): luci-app-transmission: grant access to group and passwd files

This is required for the user and group ID dropdowns in the
configuration.

Fixes: #4697
commit dda30fe7 (2021-09-24): luci-app-transmission: grant access to group and passwd files

This is required for the user and group ID dropdowns in the
configuration.

Fixes: #4697

(cherry picked from commit 1d9067b2715eede8c0e6977f9f1cc77037ff2842)