openwrt/luci #4699: luci-app-advanced-reboot for Linksys WRT1900ACS no work

liveviiv, 2020-12-30:
this Version
luci-app-advanced-reboot | git-20.343.54716-6fc0..-42

LuCI Master (git-20.343.54716-6fc079f)
--
kernel | 5.4.85

page display Warning: the device (Linksys, wrt1900acs) is unknown or not a dual partition device!

hnyman, 2020-12-30:
cc @stangri

stangri, 2020-12-30:
Hello @liveviiv -- which OpenWrt version is that?

liveviiv, 2020-12-31:
@stangri  thank you 
 https://github.com/coolsnowwolf/lede  
Is the source code compiled
I don't know how to find the relevant logs to submit to you

[image]

hnyman, 2020-12-31:
Note that your system is not normal OpenWrt. 
Based on his Chinese site, the build also includes his personal packages.
Possibly something has been changed in DTS for the in the device IDs.

liveviiv, 2020-12-31:
@hnyman 
OK, thank you very much. I'll go to / coolsnowwolf / Lede to submit issues 
By the way, where does the device ID come from.I tried to fix it myself

hnyman, 2020-12-31:
> > Warning: the device (Linksys, wrt1900acs) is unknown or not a dual partition device!

> where does the device ID come from

It comes from the DTS (devicetree).

The accepted strings are defined in the image definition:
https://github.com/openwrt/openwrt/blob/master/target/linux/mvebu/image/cortexa9.mk#L108
and then separately for the LuCI app:
https://github.com/openwrt/luci/blob/master/applications/luci-app-advanced-reboot/root/usr/share/advanced-reboot/devices/linksys-wrt1900acs.json

but the original definition comes from:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm/boot/dts/armada-385-linksys-shelby.dts?h=v5.4.86

and is modified be:
https://github.com/openwrt/openwrt/blob/master/target/linux/mvebu/patches-5.4/311-adjust-compatible-for-linksys.patch#L64

The correct one is "linksys,wrt1900acs" while your Warning example seems to show "Linksys, wrt1900acs". Note the capital "L" and space after comma. Possibly that is a typo in your first message, but possibly not. Possibly it is some kind of mistake of the author of your special firmware.

liveviiv, 2020-12-31:
@hnyman @stangri  I'll try. Thank you and I wish you a happy new year. for 2021

stangri, 2021-02-15:
Closing due to lack of activity.

closed 2021-02-15 by stangri