openwrt/luci #4700: illegal character U+0005 and sysFeatures is null

Wahjong1974, 2020-12-30:
Hi,

Got several errors after logging in this morning. 

**SyntaxError**
illegal character U+0005
  in http://*/luci-static/resources/form.js?v=git-20.363.66498-eb2cc80:53
  at compileClass/< (http://*/luci-static/resources/luci.js?v=git-20.363.66498-eb2cc80:178:29)

**TypeError**
sysFeatures is null

Strange errors. Never saw them before. You might want to take a look at it.

[image]

jow-, 2020-12-30:
Looks as if the HTTP backend is replying with garbage data. Is your flash corrupted maybe?

Wahjong1974, 2020-12-30:
I rebooted the router and the message disappeared. Really a weird error.

closed 2020-12-30 by Wahjong1974