openwrt/luci #4705: nslookup PermissionError: Permission denied

lucenera, 2021-01-01:
I am using LuCI on Turris Omnia with Turris OS 5.1.4. The LuCI version is as follows:
> Powered by LuCI branch (git-20.313.59566-7460e9d)

If I try to run nslookup in Network / Diagnostics, I get the following error:
[image]

What can it derive from? How do I get a log of what's going on behind the scenes?
If I run nslookup from the command line I have no problems and everything works as expected.

hnyman, 2021-01-01:
Works for me in normal OpenWrt (master, r15383-e1851720f1 from yesterday), even with the same address as  you used.

[image]

You might need to ask from Turris people, as you are running  their modifications.
Possibly their build has a problem with the ACL permissions.
Possibly something wrong with `/usr/share/rpcd/acl.d/luci-mod-network.json` (assuming that TurrisOS version corresponds to normal OpenWrt master).

 https://github.com/openwrt/luci/blob/master/modules/luci-mod-network/root/usr/share/rpcd/acl.d/luci-mod-network.json#L45-L62

closed 2021-01-01 by hnyman

lucenera, 2021-01-01:
Ok, I solved the problem, but i don't know yet if it depends on Turris OS or LuCI. In the meantime, the __luci-mod-network.json__ file does not exist on Turris OS, but is all integrated in __luci-base.json__ (https://termbin.com/2uom).
The file in question references the nslookup command in the _/usr/bin_ folder. But the file there is missing. So I copied the file from _/bin/_ to _/usr/bin_ and assigned the permissions provided by the other commands that are correctly executed in **Diagnostics.**
Now everything works as expected.

hnyman, 2021-01-01:
> luci-mod-network.json file does not exist on Turris OS, but is all integrated in luci-base.json 

That sounds like things were 9 months ago, before commit c099344013

lucenera, 2021-01-01:
Strange, because the commit of the version I use dates back to November 8 (7460e9d), I believe it includes all previous commits.

BKPepe, 2021-01-01:
This is caused that busybox on Turris OS installs things into `/usr` instead of `/usr/bin` and LuCI does not count with that as there are hardcoded paths. 

Nitpick:
Turris OS 5.1.4 is based on top of OpenWrt 19.07 and it was built from the followings commits: https://repo.turris.cz/archive/5.1.4/omnia/packages/git-hash

> (assuming that TurrisOS version corresponds to normal OpenWrt master).

Turris OS 6.0 is using OpenWrt master builds, but that's  branch HBD and it changes usually every day if there are no issues