openwrt/luci #4721: Wireless graph does not work

abajk, 2021-01-09:
Router: TL-WDR4300
LuCI: Master (git-21.008.63257-bc121a5)
OpenWrt: SNAPSHOT r15466-2e590a6364 

It always shows zero.

[image]

hnyman, 2021-01-09:
@jow- 
FYI, likely an additional piece broken by the iwinfo changes.

I tested with ath10k ipq806x/R7800 and mwlwifi mvebu/WRT3200ACM and the wireless realtime remains empty.

hnyman, 2021-01-09:
Likely here the libiwinfo needs modernising, too:

https://github.com/openwrt/luci/blob/master/modules/luci-mod-status/src/luci-bwc.c#L256

```
static void * iw_open(void)
{
	return dlopen("/usr/lib/libiwinfo.so", RTLD_LAZY);
}

```

commit cb454de3 (2021-01-09): luci-mod-status: realtime graph - find suffixed libiwinfo.so

Change the libiwinfo filename handling to match the current
version suffixed .so file.

Copy solution from bc121a5a4

Fixes #4721
closed 2021-01-09 by hnyman