openwrt/luci #4728: RPC call to uci/get failed with ubus code 9: Unspecified error

jendib, 2021-01-13:
Hello,

When I go to Network > Interfaces and Network > Static Routes (only those pages), I get the following error:
```
RPCError

RPC call to uci/get failed with ubus code 9: Unspecified error
  at handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-21.009.66148-9e4391e:15:3)
  at promise callback*parseCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-21.009.66148-9e4391e:9:70)
  at promise callback*call (https://192.168.1.1/luci-static/resources/rpc.js?v=git-21.009.66148-9e4391e:5:110)
  at declare/
``` 

Version: LuCI Master (git-21.009.66148-9e4391e) / OpenWrt SNAPSHOT r15477-c578fdfc29

zhihuiyuze, 2021-01-23:
[image]
`root@OpenWrt:/usr/share/rpcd/acl.d# root@OpenWrt:/usr/share/rpcd/acl.d# ls
luci-app-firewall.json  luci-app-wol.json       luci-mod-network.json   luci-wireguard.json
luci-app-opkg.json      luci-base.json          luci-mod-status.json    unauthenticated.json
luci-app-uhttpd.json    luci-compat.json        luci-mod-system.json`

[https://github.com/openwrt/luci/blob/master/modules/luci-mod-network/root/usr/share/rpcd/acl.d/luci-mod-network.json](url)

Configuration file follow here

froman753, 2021-01-26:
I too have ran into this issue when trying to access the DHCP server settings under the LAN interface tab, which states the DHCP server is not setup. 

Powered by LuCI Master (git-20.144.63033-62ed4e6) / OpenWrt SNAPSHOT r13342-e35e40ad82 
`An error occurred while saving the form:

RPC call to uci/add failed with ubus code 9: Unspecified error
  at handleCallReply (https://192.168.1.1/luci-static/resources/rpc.js?v=git-20.144.63033-62ed4e6:16:3)
`

Ansuel, 2021-01-29:
uhttpd or nginx?

jendib, 2021-01-29:
I found the issue. It was actually a badly formatted config file on my side which was causing a parse error. I found out by using the uci get command directly to get more infos.

closed 2021-01-29 by jendib

wm918, 2021-03-28:
安装了frpc，但是进服务的时候提示如下
RPCError
RPC call to uci/get failed with ubus code 4: 未找到资源
  at ClassConstructor.handleCallReply (http://192.168.2.1/luci-static/resources/rpc.js?v=git-21.069.01495-fcf7bd1:15:3)

alive-one, 2024-12-27:
> I found the issue. It was actually a badly formatted config file on my side which was causing a parse error. I found out by using the uci get command directly to get more infos.

I have same issue. I see no problems in formatting when look in vim. How you fix that?