openwrt/luci #4736: luci-app-advanced-reboot: Not rebooting to alt_kernel, EA7300v2

arrmo, 2021-01-18:
Hi,

Pulling my hair out, and I can't afford that ... LOL. I have serial access on my router, and can see the environment variables in U-Boot (and they are a bit different than in OpenWrt?). But that aside ... if I confirm / force boot_part to 1 or 2, it always seems to boot from 1. And even with a command line of,
```
bootargs=console=ttyS1,115200n8 root=/dev/mtdblock8 ro rootfstype=jffs2 init=/sbin/init
```

I still find that,
```
root@OpenWrt:/# dmesg | grep mtd
[    2.152105] UBI: auto-attach mtd6
[    2.155468] ubi0: attaching mtd6
[    2.547477] ubi0: attached mtd6 (name "ubi", size 36 MiB)
```

Thinking it may be me doing something dump, but I admit - not being able to see what it is 😞.

Thanks!

stangri, 2021-01-18:
Which version of OpenWrt is it? Please confirm is it just the Luci app, or can you not switch active partition from CLI either?

arrmo, 2021-01-18:
Hi,

Sorry! I wasn't fully clear - my fault 😞. Some answers,
- I am running master (pulled the latest from git, updated the feeds, built clean). If it helps, it's OpenWrt SNAPSHOT r14443+1087-f1bc66c765.
- It doesn't work from the LuCI app, just like you mentioned. Is there a way to manually execute it? I can try that, just not aware of it.
- The notes above are me trying to manually do it, via U-Boot. It may work fine, could just be me 😉.

Thanks!

hnyman, 2021-01-18:
>  running master (pulled the latest from git
> ...
> OpenWrt SNAPSHOT r14443+1087

Hmmm. The above claim does not match the version string.
current version should be approx. `r15535-f559b89bd0`

`r14443+1087` means 10 September 2020 code + 1087 private commits after that.

Possibly you have committed some own stuff in September but haven't properly used "git pull --rebase" after that and have just somehow merged upstream master leading to a misdetected version string.  But alternatively, that could also indicate that you do not have the features that the current LuCI app expects.  (You probably need to rebase your own commits on top of the current HEAD of OpenWrt master to get back to proper versioning. )

Likely you just have somehow screwed git repo and valid code, so this is pure nitpicking, but I still raise the point that it is hard to say what kind of OpenWrt master you are actually running.

arrmo, 2021-01-18:
You are likely right! There is a command I need ... git-idiot. Specifically for me 🤣.

I have been merging - no local changes really, but I can try a rebase. NP at all. Let me do that, see if it helps. Thanks!

arrmo, 2021-01-18:
FYI, did a `git pull --rebase upstream`, but no big delta 😞. Version is slightly changed (since yesterday, makes sense), but still,
```
r14443+1095-f1bc66c765
```

So not thinking that's the issue?

hnyman, 2021-01-18:
You already have earlier merges there, so the "new" git pull rebase does not help any more as it only tries the newest stuff.

I think that you need something like "git rebase upstream/master" (or "git rebase origin/master", depending on your git config).

arrmo, 2021-01-18:
Looks like I was "fooling" getver.sh 😄. I was pulling from upstream (no issue there), but had been lazy at pushing back to my fork (on Github). So I just did a `git push origin master`, no code changes at all, and now getver.sh says,
```
r15538-26d1e529f1
```

So I think the code I'm running really is current, just having the issue noted above.

Is there a CLI way to try to execute advanced-reboot? Or U-Boot commands to try? Thanks!

arrmo, 2021-01-19:
OK, this is really odd. Tried setting boot_part inside OpenWrt, rebooting ... nope, doesn't seem to switch partitions. Either value for boot_part and I see in the kernel log,
```
[    2.151726] UBI: auto-attach mtd6
[    2.160787] ubi0: attaching mtd6
```

FYI, in case it helps / matters,
```
cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "boot"
mtd1: 00040000 00020000 "u_env"
mtd2: 00040000 00020000 "factory"
mtd3: 00040000 00020000 "s_env"
mtd4: 00040000 00020000 "devinfo"
mtd5: 00400000 00020000 "kernel"
mtd6: 02400000 00020000 "ubi"
mtd7: 00400000 00020000 "alt_kernel"
mtd8: 02400000 00020000 "alt_rootfs"
mtd9: 00100000 00020000 "sysdiag"
mtd10: 02d00000 00020000 "syscfg"
```

I also tried setting the root partition inside U-Boot (tried mtdblock7 and 8),
```
setenv bootargs console=ttyS1,115200n8 root=/dev/mtdblock7 ro rootfstype=jffs2 init=/sbin/init
```
But then, inside OpenWrt, with `fw_printenv`,
```
bootargs=console=ttyS1,115200n8 root=/dev/mtdblock6 ro rootfstype=jffs2 init=/sbin/init
```

Huh? Seems to overwrite the bootargs (on boot)? I admit, very confusing.

Thanks!

stangri, 2021-01-19:
AFAIK you also need to modify `bootcmd` in sync with `boot_part`.

It is also possible that alternative partition is corrupt in some way, so router reverts to previously successfully booted partition after 3 failed attempts at booting from alternative.

Either way, it doesn't seem to be the luci app-specific problem, seems to be more of a generic problem with rebooting to alternative partition on your router. If you get the CLI commands to work to reboot to alt partition, but the luci app still doesn't, please re-open this issue.

closed 2021-01-19 by stangri

arrmo, 2021-01-19:
Yes, agreed - appreciate the comments. If you do have any pointers, by all means 😆.

Thanks!

hnyman, 2021-01-19:
> Is there a CLI way to try to execute advanced-reboot? Or U-Boot commands to try? Thanks!

I have been including in my  WRT3200ACM build also this simple script (pre-dating the LuCI app) that does the thing.
you could modify it.

```
#!/bin/sh
#
# This script changes the router to boot from the alternative kernel/partition.
#
# Note: you have to make this executable: "chmod 755 switch_boot_partition.sh"
#
# Normally partition is changed at firmware flash. New firmware is flashed
# to the other partition and then the router boots from there. This script
# causes that boot partition change without a flash

cur_boot_part=`/usr/sbin/fw_printenv -n boot_part`
if [ "$cur_boot_part" = "1" ]
then
	echo "Switching to kernel/partition 2"
	fw_setenv boot_part 2
	fw_setenv bootcmd "run altnandboot"
elif [ "$cur_boot_part" = "2" ]
then
	echo "Switching to kernel/partition 1"
	fw_setenv boot_part 1
	fw_setenv bootcmd "run nandboot"
else
	echo "Error: unexpected value: boot_part=$cur_boot_part"
fi

# auto_recovery should have been set to 'yes' by /etc/init.d/97linksys_recovery
# the same script has also reseted the boot-counter used by u-boot.

fw_printenv auto_recovery
```
That is copied and modified from the original used in OpenWrt...
https://github.com/openwrt/openwrt/blob/master/target/linux/mvebu/cortexa9/base-files/lib/upgrade/linksys.sh

stangri, 2021-01-19:
@hnyman Thanks to @Ansuel with the new luci app you can `ubus call luci.advanced_reboot toggle_boot_partition`. ;)

hnyman, 2021-01-19:
Nice.

But the point is that when he is having major problems in getting the partition switch to happen, it is better to see the actual underlying steps and maybe execute them by hand, instead of just using your wrapper that hides the actual happenings.

arrmo, 2021-01-19:
Agree wholeheartedly with both of you! That's what I want as well - want to make sure to understand the underlying issue here, ensure the solution is 100% clean!

It's seeming like boot_part isn't the only (environment) variable in play - but I may be wrong. I want to make sure I have two clean partitions, to confirm. I have checked my builds, can run them from U-Boot (tftpboot to load the initramfs file, then bootm). Struggling to get the factory.bin to program to the flash. The "nand write" command info is a bit cryptic, don't want to guess to much, or I risk the full flash ... 😞.

Thanks again for the pointers!

arrmo, 2021-01-19:
BTW, comparing this device to the EA3500 I have, I get the feeling that early builds that I flashed messed up the u_env partition, that may be part of the challenge here. For example, there is no altnandboot command defined.