openwrt/luci #4744: Luci-app-sqm won’t automatically start after boot(static address)

rulet, 2021-01-20:
sqm rules doesn't work after reboot, I have to manually restart it, openwrt 19.07.5

feckert, 2021-01-20:
The package https://github.com/openwrt/luci/tree/openwrt-19.07/applications/luci-app-sqm is not presented on openwrt-19.07 I can not find any sqm in the 19.07 branch. The acl json must be an artifact from a backport. Please check the package feed https://github.com/openwrt/packages/blob/openwrt-19.07/net/sqm-scripts/Makefile

rulet, 2021-01-20:
https://openwrt.org/packages/pkgdata/luci-app-sqm

rulet, 2021-01-20:
I fixed the issue by workaround [described here](https://forum.openwrt.org/t/how-to-keep-luci-app-sqm-started-and-enabled-when-rebooting-the-router/8019/6?u=rulet)

feckert, 2021-01-20:
Again, this is not a LuCI bug because the packet does not exist in openwrt 19.07 branch according to my research. Since the package is maintained in the package feed you have to open an issue there.
https://github.com/openwrt/packages

hnyman, 2021-01-20:
> not a LuCI bug because the packet does not exist in openwrt 19.07 branch 

It does exist, but in the packages repo...
(tohojo originally defined it strangely there, and that has been changed for master in April by https://github.com/openwrt/packages/commit/3e46fd5f05104d169730ebb1091a2ac322a9e341)

https://github.com/openwrt/packages/blob/openwrt-19.07/net/sqm-scripts/Makefile#L48

But the LuCI app has no role in the actual startup of the underlying SQM package itself, so the "SQM not enabled at startup" bug does not belong here in LuCI.

closed 2021-01-20 by hnyman

feckert, 2021-01-20:
@hnyman should we not delete the luci-app-sqm in luci on branch openwrt-19.07 as i said?
This must be an artifact during a merge.
The acl file does not make sense and on the [package side](https://openwrt.org/packages/pkgdata/luci-app-sqm ) you can wrongly see that we have a sqm app on branch 19.07.

rulet, 2021-01-20:
> bug does not belong here in LuCI

So, to whom I have to talk?

hnyman, 2021-01-20:
cc @jow- 

> you can wrongly see that we have a sqm app on branch 19.07 but this is not the case.

We do have the LuCI app in 19.07, like I linked above:
https://github.com/openwrt/packages/blob/openwrt-19.07/net/sqm-scripts/Makefile#L48
```
define Package/luci-app-sqm
  SECTION:=luci
  CATEGORY:=LuCI
  TITLE:=SQM Scripts - LuCI interface
```

And it is downloadable:
https://downloads.openwrt.org/releases/19.07.6/packages/arm_cortex-a15_neon-vfpv4/packages/luci-app-sqm_1.4.0-2_all.ipk

It is just defined, compiled and downloaded from the packages feed, not LuCI.

For master it was moved properly to the LuCI feed after discussion in 
https://forum.openwrt.org/t/translate-luci-app-sqm/59131

But for 19.07 it has been left in the old style of being defined in the packages feed, but actually being sourced in tohojo's own repo.

Like I said, sqm (and acme) were anomalies, as also the LuCI apps were defined in upstream (in tohojo's repo)

But that ACL definition file in the LuCI 19.07 repo is wrong/unutilised, as it naturally does not get included in the luci-app-sqm build done from the packages feed.  (@jow- had not figured that anomaly out in the backport that he has done with ecd4924)

I am not sure if that ACL is actually needed in 19.07, but if it is needed, it should be then defined in the packages repo.

Or maybe the luci-app-sqm should also be defined normally in the LuCI repo.

hnyman, 2021-01-20:
@rulet 

>  to whom I have to talk?

Errors/bug for the SQM package itself are best discussed with @tohojo or @moeller0 in either
https://github.com/openwrt/packages/issues or
https://github.com/tohojo/sqm-scripts/issues

feckert, 2021-01-21:
@hnyman Sorry for the confusion from my side! I did not see it that the luci-app-sqm is maintained in the package feed. It is different in the master branch now. I was irritated that there is the directory in the LuCI feed with only the acl file. So I assumed it was a mistake. I have now staged a [P/R](https://github.com/TDT-AG/luci/tree/pr/20210121-luci-app-sqm) in my repository that removes the acl, because (as I see it) this is not necessary for the openwrt-19.07 branch.

@rulet In openwrt there is the LuCI feed, package feed and openwrt base and some other feeds. If the problem is about the WebGUI then an issue should be raised here. If it is an issue in the system (like in your case with sqm), then an issue should be raised in the [package feed](https://github.com/openwrt/packages/issues).

Just a hint from me to solve the problem for sqm.
I think you would have to restart the sqm on an interface event.
For this you have to look at the directory /etc/hotplug.d/iface place a script there that do the job.

moeller0, 2021-01-21:
Hi Florian,

so currently, sqm has the following hotplug script:

[ -n "$DEVICE" ] || exit 0

restart_sqm() {
    /usr/lib/sqm/run.sh stop ${DEVICE}
    /usr/lib/sqm/run.sh start ${DEVICE}
}

[ "$ACTION" = ifup ] && /etc/init.d/sqm enabled && restart_sqm

[ "$ACTION" = ifdown ] && /usr/lib/sqm/run.sh stop ${DEVICE}

which seems to work wel for e.g. pppoe-wan coning and going. What specifically do we need to add for static IP interfaces?

Best Regards
	Sebastian

feckert, 2021-01-21:
@moeller0 Sorry but I can't find your posted script anywhere in the package feed repo!
Is this checked in or is this just a suggestion for the issue.
As I said I have now not dealt with sqm more closely.
Just wanted to be helpful in the LuCI.

moeller0, 2021-01-21:
Hi Florian,

> 
> 
> @moeller0 Sorry but I can't find your posted script anywhere in the package feed repo!

	Oh, it is part of the main sqm-scripts repository see https://github.com/tohojo/sqm-scripts/blob/master/platform/openwrt/sqm-hotplug as it is not specific to the luci GUI, but rather required for the main packet to function in light of appearing/disappearing interfaces/devices.

> Is this checked in or is this just a suggestion for the issue.

	Yes, that comes as part of the sqm-scripts package, I copied the code verbatim from /etc/hotplug.d/iface/11-sqm  

> As I said I have now not dealt with sqm more closely.
> Just wanted to be helpful in the LuCI.

	Thanks, much appreciated. I might well be that our script is confused in regards to the subtle? differences between $INTERFACE and $DEVICE, any information/documentation would be appreciated.

Best Regards
	Sebastian

feckert, 2021-01-21:
@moeller0 
> Oh, it is part of the main sqm-scripts repository see https://github.com/tohojo/sqm-scripts/blob/master/platform/openwrt/sqm-hotplug as it is not specific to the luci GUI, but rather required for the main packet to function in light of appearing/disappearing interfaces/devices.

Now I have found it too. Thanks for the hint

> Thanks, much appreciated. I might well be that our script is confused in regards to the subtle? differences between $INTERFACE and $DEVICE, any information/documentation would be appreciated. Best Regards Sebastian

I found [this](https://openwrt.org/docs/guide-user/base-system/hotplug). Which explains the variables better.

Maybe we have to change the handling, So we do not specify the device (eth1, pppoe-xdsl..) but rather the logical openwrt interface (wan, xdsl, wlan). The advantage would be we would be independent of the correct device name, that would simplify the matter. We would only reference the logical openwrt name and get the real hardware device at runtime.

I would then wrap it all up in the init script so we don't have to change the run script. That means we could turn sqm on at startup. The interfaces that are already there would start up and the others that start later (xdsl, wwan) would be started via the hotplug system. They could call then init for example **/etc/init.d/sqm up/down <interface>**

rulet, 2021-01-21:
So, if I understand right, the bug exists. When to expect for patch, I mean when in stable repo the sqm package will be updated? For now I'm using workaround which is [described here](https://forum.openwrt.org/t/how-to-keep-luci-app-sqm-started-and-enabled-when-rebooting-the-router/8019/6?u=rulet), but feckert says to place a script in /etc/hotplug.d/iface. How exactly that script should look like, can you give an example?

moeller0, 2021-01-21:
> 
> 
> So, if I understand right, the bug exists.

	Not 100% sure what the bug is.

> When to expect for patch,

	Not before someone created such a patch/fix ;)

> I mean when in stable repo the sqm package will be updated?

	Well, before the issue is fully diagnosed and a fix created and confirmed to actually work no updates are going to happen.

> For now I'm using workaround which is described here, but feckert says to place a script in /etc/hotplug.d/iface. How exactly that script should look like, can you give an example?

	The thing is there already is a script that as far as I can tell from the documentation should work. How about we take this up on this weekend again and create an instrumented version of the sqm hotplug script to figure out why it does not trigger?

Best Regards
	Sebastian

rulet, 2021-01-21:
> How about we take this up on this weekend again and create an instrumented version of the sqm hotplug script to figure out why it does not trigger?

Ok, I can test it on my device(Archer c6) if it's possible.

patrakov, 2021-01-21:
We did a private debugging session with @rulet and found this.

The `sqm` initscript is enabled correctly.

However, there is a strange network config (a consequence of watching a possibly-misleading tutorial at https://www.youtube.com/watch?v=7982Z44ebOc):

```
config interface 'lan'
        option type 'bridge'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option ifname 'eth0.1 eth0.3'

...

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option vid '1'
        option ports '0t 2 5'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '0t 1'
        option vid '2'

config switch_vlan
        option device 'switch0'
        option vlan '3'
        option ports '0t 3 4'
        option vid '3'
```

I.e. the bridging between eth0.1 and eth0.3 is done in software, so that SQM can only be applied to one part of the LAN.

The SQM config is:

```
config queue
        option debug_logging '0'
        option verbosity '5'
        option enabled '1'
        option interface 'eth0.1'
        option qdisc_advanced '0'
        option download '70000'
        option upload '70000'
        option linklayer 'ethernet'
        option overhead '44'
        option qdisc 'cake'
        option script 'piece_of_cake.qos'

config queue
        option debug_logging '0'
        option verbosity '5'
        option enabled '1'
        option interface 'eth0.3'
        option download '0'
        option upload '0'
        option qdisc_advanced '0'
        option linklayer 'ethernet'
        option overhead '44'
        option qdisc 'cake'
        option script 'piece_of_cake.qos'
```

We have modified /etc/hotplug.d/iface/11-sqm as follows:

```
#!/bin/sh
exec >>/tmp/hotplug.log 2>&1
date
env
set -x

[ -n "$DEVICE" ] || exit 0

restart_sqm() {
    /usr/lib/sqm/run.sh stop ${DEVICE}
    /usr/lib/sqm/run.sh start ${DEVICE}
}

[ "$ACTION" = ifup ] && /etc/init.d/sqm enabled && restart_sqm

[ "$ACTION" = ifdown ] && /usr/lib/sqm/run.sh stop ${DEVICE}
```

Here is what has been logged after a reboot:

```
Thu Jan 21 13:20:11 UTC 2021
USER=root
ACTION=ifup
SHLVL=1
HOME=/
HOTPLUG_TYPE=iface
LOGNAME=root
DEVICENAME=
TERM=linux
PATH=/usr/sbin:/usr/bin:/sbin:/bin
INTERFACE=lan
PWD=/
DEVICE=br-lan
+ '[' -n br-lan ]
+ '[' ifup '=' ifup ]
+ /etc/init.d/sqm enabled
+ restart_sqm
+ /usr/lib/sqm/run.sh stop br-lan
+ /usr/lib/sqm/run.sh start br-lan
+ '[' ifup '=' ifdown ]
Thu Jan 21 13:20:12 UTC 2021
USER=root
ACTION=ifup
SHLVL=1
HOME=/
HOTPLUG_TYPE=iface
LOGNAME=root
DEVICENAME=
TERM=linux
PATH=/usr/sbin:/usr/bin:/sbin:/bin
INTERFACE=loopback
PWD=/
DEVICE=lo
+ '[' -n lo ]
+ '[' ifup '=' ifup ]
+ /etc/init.d/sqm enabled
+ restart_sqm
+ /usr/lib/sqm/run.sh stop lo
+ /usr/lib/sqm/run.sh start lo
+ '[' ifup '=' ifdown ]
Thu Jan 21 13:20:14 UTC 2021
USER=root
ACTION=ifup
SHLVL=1
HOME=/
HOTPLUG_TYPE=iface
LOGNAME=root
DEVICENAME=
TERM=linux
PATH=/usr/sbin:/usr/bin:/sbin:/bin
INTERFACE=wan
PWD=/
DEVICE=eth0.2
+ '[' -n eth0.2 ]
+ '[' ifup '=' ifup ]
+ /etc/init.d/sqm enabled
+ restart_sqm
+ /usr/lib/sqm/run.sh stop eth0.2
+ /usr/lib/sqm/run.sh start eth0.2
+ '[' ifup '=' ifdown ]
```

See - the hotplug event is about `br-lan`, while the config is about `eth0.1` and `eth0.3`. OTOH I am not sure if the use case (limiting the speed of a bridge port) is valid at all.