openwrt/luci #4747: Luci wireguard

predators46, 2021-01-26:
when does openwrt have a wireguard luci that can import accounts like glinet

feckert, 2021-01-26:
So far, there are no plans to do so. Besides, you need to give us more information about what it is. Best with pictures and a link where we can look at it, what this exactly is

predators46, 2021-01-26:
> So far, there are no plans to do so. Besides, you need to give us more information about what it is. Best with pictures and a link where we can look at it, what this exactly is

https://docs.gl-inet.com/en/3/app/wireguard/

predators46, 2021-01-26:
[image]
[image]

predators46, 2021-01-26:
I use glinet and use glinet firmware I really like the wireguard account import feature I hope the openwrt devoloper can make it like that

predators46, 2021-02-17:
> So far, there are no plans to do so. Besides, you need to give us more information about what it is. Best with pictures and a link where we can look at it, what this exactly is

hy. I tried the package file which must overwrite the existing config file. When I installed the file I found an error check_data_file_clashes how can I overwrite an existing config file

jow-, 2022-12-19:
Current master provides import and export facilities for WireGuard configs.

closed 2022-12-19 by jow-