openwrt/luci #4758: /usr/lib/lua/luci/dispatcher.lua:370: /etc/config/luci seems to be corrupt, unable to find section 'main'

rozhuk-im, 2021-02-02:
Update to latest master and restore config from backup (~6 month old), got error:
```/usr/lib/lua/luci/dispatcher.lua:370: /etc/config/luci seems to be corrupt, unable to find section 'main'```

#2885 and #2231 - this is not first time.

Workaroud in this message: https://github.com/openwrt/luci/issues/2231#issuecomment-663915276
work for me: if I run ```rpcd``` from ssh then error gone.

I dig inside and found that in my backup /etc/config/rpcd
```
config rpcd
	option socket /var/run/ubus.sock
	option timeout 30

config login
	option username 'root'
	option password '$p$root'
	list read '*'
	list write '*'
```

This is from default config after reset:
```
config rpcd
	option socket /var/run/ubus/ubus.sock
	option timeout 30

config login
	option username 'root'
	option password '$p$root'
	list read '*'
	list write '*'
```

As you can see:
1. ```option socket``` is different.
2. if ```rpcd``` started without options - if uses proper path to socket, at least GUI works.

Why not remove ```option socket``` handling from ```/etc/config/rpcd``` and ```/etc/init.d/rpcd``` ?

jow-, 2021-02-09:
Not a LuCI bug. Your config need to be updated.

closed 2021-02-09 by jow-

mmokrejs, 2021-06-26:
```
/usr/lib/lua/luci/dispatcher.lua:427: /etc/config/luci seems to be corrupt, unable to find section 'main'
```

Same issue after upgrade from 19.07 to `Powered by LuCI openwrt-21.02 branch (git-21.163.64918-ba57ec5) / OpenWrt 21.02.0-rc3 r16172-2aba3e9784`:

```
# cat /etc/config/rpcd
config rpcd
	option socket /var/run/ubus.sock
	option timeout 30

config login
	option username 'root'
	option password '$p$root'
	list read '*'
	list write '*'

#
```

```
# ls -latr /var/run/ubus.sock
ls: /var/run/ubus.sock: No such file or directory
# ls -latr /var/run/ubus/
srw-rw-rw-    1 root     root             0 Jan  1  1970 ubus.sock
drwxr-xr-x    2 root     root            60 Jan  1  1970 .
drwxr-xr-x    7 root     root           420 Jun 26 02:26 ..
# 
```

HubKing, 2021-09-06:
> /var/run/ubus/ubus.sock

I upgraded to 21.02 and this happened. changing `/var/run/ubus.sock` to ` /var/run/ubus/ubus.sock` fixed it, but isn't this a bug? I don't know who is to blame, but either the OpenWRT or Luci needs to fix this, so that users don't have to do this manually.

Cadair, 2021-11-27:
I too got this after using the real time graphs.

n-pat, 2022-03-02:
I can confirm that I also got this when using real time graphs. I could solve by manually adjusting the ubus path in the rpcd config.

mmokrejs, 2022-03-02:
@n-pat Please include release version numbers in your above answer. Thanks.

exaghost, 2022-03-07:
Hi!
I have the same issue. To reproduce: login to Luci, and browse the graphs. I also have many connections. Then the corruption occurs. To fix it, I can restart the service over SSH (service rpcd restart), and everything works normal. This problem exists in 21.02.0 (fresh install) and 21.02.1. (upgrade) I think it's not a Luci bug.

davidtwgit, 2022-03-14:
Same issue when I flash 19.x from luci on 21.02.2  with router r4ac 100m

IOsetting, 2022-04-10:
> I can confirm that I also got this when using real time graphs of connections (with about 5000 open connections). rpcd seems to crash reproducable. A restart of rpcd helps.
> 

I got this same error, after visiting the connections page of real time graphs.

beadon, 2022-08-22:
Reporting that I also have the same problem with a large number of connections, crash happens after enabling the DNS resolution in the connections display via LUCI web interface.

mmokrejs, 2022-08-22:
> I also encountered this symtom after attempting to view the "Connections" real-time graphs, and a restart of rpcd made it work again.
> 
> Although there's apparently an underlying issue with rpcd, it seems that the error presented by Luci is bogus - it suggests that the 'main' section is missing from /etc/config/luci, which is, in fact, not the case. From reading above, it appears that this error is also presented when an incorrect rpcd socket path is configured. Apparently this is actually a general "oh crap, something bad happened [when trying to talk to rpcd?]" condition. The bogus error message seems like a Luci bug, albeit a somewhat cosmetic one.

Well, maybe in your case it was cosmetis issue but in my case it was a fatal error. The path to the socket was just wrong causing rpcd to exit. But I agree the other situations should be covered a different, more descriptive error message.

HubKing, 2022-09-25:
I bought a new router yesterday. After uploading the initial ROM to convert stock->OpenWRT, I straightly installed the latest OpenWRT. I am not sure if that counts as an upgrade or not, but this problem randomly happened today. That is, OpenWRT had been working find for hours and when I was browsing the real-time graph tab,s it happened

    /usr/lib/lua/luci/dispatcher.lua:431: /etc/config/luci seems to be corrupt, unable to find section 'main'

This time,  there was no `/var/run/ubus.sock`  and `/var/run/ubus/ubus.sock`  was already there, so I am not sure sure how to solve this.

mmokrejs, 2022-09-25:
@HubKing Please include above which OpenWRT version you installed and filname if of imge you flashed.

HubKing, 2022-09-25:
> Please include above which OpenWRT version you installed and filname if of imge you flashed.

- Version: OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a
- Filename: openwrt-22.03.0-mediatek-mt7622-xiaomi_redmi-router-ax6s-squashfs-sysupgrade.bin

dseven, 2023-04-27:
IMO there should be two separate new issues opened for this problem:

1) RPC dies when there are lots of connections (not a Luci issue, so won't he addressed here)
2) Luci presents a misleading error message when RPC dies (this is a Luci issue, IMO)

emhamm, 2023-04-27:
You are right, i will open those issues now.