openwrt/luci #4760: WRT1200AC 19.07.6 Luci issue

termathor, 2021-02-03:
Hello,

Just to report a tiny issue, easy to reproduce, whatever the config is:

- in Luci, set up Size of DNS query cache to whatever different from default
- restart the router
- > DNSmasq will fail to boot since the generated config file has unknown keyword

hnyman, 2021-02-03:
Sounds dubious.
I have been using cache size 1000 for years, and just changed it via LuCI to 1001, and the dsnmasq service reloaded the new config just fine.  

I haven't tested changing it from LuCI in 19.07 recently, but I fail to see how there could be anything dangerous there.

Note that LuCI only writes to the uci config file /etc/config/dhcp, and the dnsmasq service then parses it when it launches (at boot, or otherwise).

jow-, 2021-02-17:
Closing this as it appears to be unrelated to LuCI.

closed 2021-02-17 by jow-