openwrt/luci #4761: .depends() for key not being used in a LuCI form

nickberry17, 2021-02-04:
Hey,

I would like to use the [.depends()](http://openwrt.github.io/luci/jsapi/LuCI.form.Value.html#depends) method to conditionally render the LuCI form option based on the value of a key that is not being used or displayed in the form.

From the documentation: 
> Dependency constraints allow making the presence of option elements dependant on the current values of certain other options _within the same form_.

My scenario, in which `pukrequired` is just set by the protocol handler on the fly, and never really needs to be seen or changed by the user.
```
o = s.taboption('sim1', form.Value, 'puk', _('PUK (Personal Unlocking Key)'),
		_("Your SIM requires a PUK to unlock it."));
o.depends("pukrequired", "yes" );
```

Is it possible for this to be implemented?  If so, could someone please point me to the relevant code?  If not, any comments on potential workarounds for such a scenario?

hnyman, 2021-02-04:
I think that you might use DummyValue for pukrequired and use it as a dependency for puk (like now).
That might be be easiest approach.

nickberry17, 2021-02-06:
Thank you, that was straightforward to get the value, although I did need to set a uci value to some HTML in proto handler because I wanted to make the dummy value completely hidden.  Bad smell.  So I added a new member called `hidden` to `DummyValue` to allow us to do this quite nicely from anywhere.  I hope this can be accepted in a MR after some review and discussion.

Before
```
o = s.taboption('general', form.DummyValue, 'pukrequired');
o.rawhtml = true;
o = s.taboption('sim1', form.Value, 'puk1', _('PUK (Personal Unlocking Key)'),
	_("Your SIM requires a PUK to unlock it."));
o.depends({ pukrequired: "<p hidden>yes</p>", simslot: "1" })
```
After:
```
o = s.taboption('general', form.DummyValue, 'pukrequired');
// the new member
o.hidden = true; 

o = s.taboption('sim1', form.Value, 'puk1', _('PUK (Personal Unlocking Key)'),
	_("Your SIM requires a PUK to unlock it."));
// no more bad smells
o.depends({ pukrequired: "yes", simslot: "1" })
```

hnyman, 2021-02-06:
>  I wanted to make the dummy value completely hidden

Why not just show that on the screen as an informative item?

nickberry17, 2021-02-06:
> > I wanted to make the dummy value completely hidden
> 
> Why not just show that on the screen as an informative item?

In this case, the mere presence of the puk field was considered to be enough information, and the other one was cluttering the UI with non-essential information.

closed 2021-02-06 by nickberry17