openwrt/luci #4788: Multiple remote LUCI

nsmith1024, 2021-02-09:
Hello,

Im new to LUCI, and i need to add some features to it and im currently trying to find the best way to do it.

To use LUCI right now the user has to log into a particular router's web interface, but if i have 100 routers i dont want to log into each router one by one, i want to log into one router, make changes, then be able to push those changes to all the other routers automatically without manually logging into all 100 routers one by one.

Another feature i need is to be able collect metrics (number of packet errors, signal strength, etc) from all the routers and see them from one router's web interface dashboard, instead of running around logging in and looking at each router individually.

Anybody has any suggestions on how to do any of these?

Thanks

hnyman, 2021-02-09:
> Another feature i need is to be able collect metrics (number of packet errors, signal strength, etc) from all the routers and see them from one router's web interface dashboard, instead of running around logging in and looking at each router individually.

That is already possible with LuCI statistics, as collectd (the underlying stat engine) supports networking.

hmh, 2021-02-15:
@nsmith1024: several openwrt downstreams implented stuff like this, but typically they are "not sharing".  We need a very barebones version of it too, so I've been looking into it.

Once the router is deployed, it is trivial to do using ubus over uhttpd + https, but you will want firewall restrictions if you enable this in the WAN, as well as ensuring proper SSL validation to avoid MITM.  Otherwise, you're toast.

Use a separate user (add it to rpcd) for ubus access, with an extremely strong password (machine-generated).  I recommend using a strong base secret, PBKDF2, and a HKDF or other securely hashed variation of the device MAC as salt for the PBKDF2, obviously *also* using a secret.  That allows per-device credentials, being accessed from a single centralized location (which holds the other secrets).

hmh, 2021-02-15:
Or just create per-device credentials normally with a string password generator, and store them in a safe database inside your centralized control server.  Whatever works best for you.

jow-, 2021-02-17:
As @hmh pointed out, using ubus over HTTP(s) would be the way to go. LuCI slowly migrates in that direction, handling more and more things on the client side so that eventually only static assets are left which could theoretically be hosted anywhere and be instructed to connect to any given ubus HTTP endpoint on login.

Right now this is not feasible yet because a number of LuCI apps are still server side rendered and some base LuCI logic (dispatching, header/footer framing etc.) still happens server side.

Closing this as this is not really an actionable bug report.

closed 2021-02-17 by jow-