openwrt/luci #4801: add missing btn class to buttons

tapper82, 2021-02-10:
Hi can we pleas add missing btn class to buttons like in this PR:
https://github.com/openwrt/luci/commit/34417400893d8513f92c4e4ba3a1b8c9207a5392

Packages I have found that need this change are:
Adblock, banip and sqm. 
Then there is some luci pages that need fixing to. 
system, interfaces, Wireless, switch, DHCP and DNS, Hostnames, Static Routes and firewall.
It's the Save & Apply button that is not working as it should.

stangri, 2021-02-11:
tagging @dibdot.

dibdot, 2021-02-11:
@tapper82 What's the purpose of this change? Better screenreader compatibility?

tapper82, 2021-02-11:
@dibdot Yes and makes the button work better with the keyboard.

dibdot, 2021-02-11:
banIP will be fixed with #4804. The adblock change has been noted for the next regular update. Thanks for bringing this up.

tapper82, 2021-02-11:
Thanks for fixing!

rmilecki, 2021-02-12:
I had a quick chat with @tapper82 and it may be better to actually use `role="button"` for button links. It seems to work the same way for screen readers but at the same time it's more explicit.

https://developer.mozilla.org/en-US/docs/Web/Accessibility/ARIA/Roles/button_role

dibdot, 2021-02-12:
@rmilecki to be honest, I don't know why a screen reader doesn't already recognize the buttons as such today, because we're using   the native \<button>...\<\button> syntax (at least in adblock/banIP), therefore I don't think we need a extra "role" here. See the #best_practices anchor in your linked document.

dibdot, 2021-03-03:
@tapper82 adblock has been fixed as well (in master & 21.02)

tapper82, 2021-03-03:
@dibdot Thanks.

closed 2022-04-12 by feckert