openwrt/luci #4806: Newbie question around Firewall/Zones

marcus-j-davies, 2021-02-12:
Hello

I have just switched to OpenWRT/Luci (from stock firmwware).
And have a question, around the firewall configuration.

My network consists of around ~80 endpoints, so my setup is 'modest'.

I have marked the image below.
The blue, I get/understand, but the red highlighted column, I need help, on - I am not an expert at iptables (I am a UI Junkie)

My question is,
Is this controlling the traffic coming  into the router, and targeted for the associated zone.
i.e is `INPUT` the traffic coming into the router (regardless of source zone), and heading to said zone
if that fact is true, does that mean `OUTPUT` is the traffic originating from that zone.

OR

is this traffic originating from the router?
i.e  `INPUT ` is traffic from router to zone, and `OUTPUT` is traffic from  zone to router.

Thanks

[image]

jow-, 2021-02-12:
- INPUT: traffic from the associated zone towards the router itself (e.g. from LAN client to router's DNSmasq or web ui)
- OUTPUT: traffic generated on router and routed towards related zone (e.g. running "ping google.com" via SSH on the router itself would hit the OUTPUT policy for wan)

closed 2021-02-12 by jow-

marcus-j-davies, 2021-02-12:
OK - That helps - thank you
Are these applied prior the the rules defined under 'Traffic Rules' i.e which takes priority?