openwrt/luci #4810: luci-app-samba4: compilation error

Ansuel, 2021-02-14:
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for libreadline8 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package libreadline8.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-samba4:
 *      libncursesw6
 * opkg_install_cmd: Cannot install package luci-app-samba4.
make[2]: *** [package/Makefile:69: package/install] Error 255

ptpt52, 2021-02-16:
same issue here:
```
Configuring ppp-mod-pppoe.
Configuring iwlwifi-firmware-iwl9260.
Configuring iptables-mod-rpfilter.
Configuring luci-proto-ncm.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for libreadline8 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package libreadline8.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for gnupg:
 * 	libncursesw6
 * opkg_install_cmd: Cannot install package gnupg.
package/Makefile:66: recipe for target 'package/install' failed
```

ptpt52, 2021-02-16:
@adschm @Ansuel seems related to openwrt/openwrt resp. changes

lengfwang, 2021-02-16:
same:
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for libreadline8 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package libreadline8.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-samba4:
 *      libncursesw6
 * opkg_install_cmd: Cannot install package luci-app-samba4.
package/Makefile:66: recipe for target 'package/install' failed

hnyman, 2021-02-16:
Please delete the temporary package data from tmp/ and update OpenWrt sources plus all package feeds, and then compile after a make clean.

There have been so many ABI & dependency changes in the last few days, that you really need to clean the buildroot thoroughly.

(And if your are talking about the imagebuilder, please wait 1-2 days, so that the buildbot has had tiem to compile all SDKs with the current (/hopefully right) code and then later all packages with that updated SDK for your architecture)

ptpt52, 2021-02-17:
test build for target x86_64, remove `build_dir tmp staging_dir bin`
and rebuild, still failed

(but no failure on target ath79)

you may reproduce it by select these package built-in
```
fdisk
sfdisk
bash
```
```
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for libreadline8 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package libreadline8.
 * pkg_hash_fetch_best_installation_candidate: Packages for fdisk found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package fdisk.
 * pkg_hash_fetch_best_installation_candidate: Packages for sfdisk found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package sfdisk.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for bash:
 * 	libncursesw6
 * opkg_install_cmd: Cannot install package bash.
```

ptpt52, 2021-02-17:
I am trying to make some changes and it build ok
```
diff --git a/package/libs/ncurses/Makefile b/package/libs/ncurses/Makefile
index cc1e960a0c..bc5ddead4b 100644
--- a/package/libs/ncurses/Makefile
+++ b/package/libs/ncurses/Makefile
@@ -38,7 +38,7 @@ define Package/libncurses
   CATEGORY:=Libraries
   TITLE:=Terminal handling library (Unicode)
   URL:=http://www.gnu.org/software/ncurses/
-  PROVIDES:=libncursesw
+  PROVIDES:=libncursesw libncursesw6
   DEPENDS:= +terminfo
   ABI_VERSION:=6
 endef

```

@hnyman @Ansuel

ptpt52, 2021-02-17:
https://github.com/openwrt/openwrt/pull/3875

hoatienii, 2021-02-18:
> I am trying to make some changes and it build ok
>
> ```
> diff --git a/package/libs/ncurses/Makefile b/package/libs/ncurses/Makefile
> index cc1e960a0c..bc5ddead4b 100644
> --- [...]

Thanks so much! Same error with luci-app-diskman. Try you solution and work fine.

avbohemen, 2021-02-18:
I had the same error yesterday, with me it was only libreadline8. It is now (probably) solved by https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=0db9d118657dd93509abec817eb22ce13d0ec1b6

closed 2021-02-19 by feckert