openwrt/luci #4812: Firewall: Cannot add rules with negative CIDR

SoniEx2, 2021-02-15:
fw3 supports dynamic prefix forwarding: https://openwrt.org/docs/guide-user/firewall/fw3_configurations/fw3_ipv6_examples#dynamic_prefix_forwarding

However, LuCI doesn't allow one to input the `/-64` syntax.

dibdot, 2021-02-16:
A negative CIDR looks invalid, most probably a typo or a range designation in the wiki.

SoniEx2, 2021-02-16:
It was in one of the release notes at some point. It's definitely not invalid.

dibdot, 2021-02-16:
The number following the slash is the prefix length ... what describes a negative number in this case? Do you have any implementation references?

SoniEx2, 2021-02-16:
It's the suffix length.

SoniEx2, 2021-02-16:
this clearly explicitly supports negative values: https://github.com/gigibox/fw3/blob/a4d98aea373e04f3fdc3c492c1688ba52ce490a9/utils.c#L771

altho ugh, the lack of comments makes it impossible to follow. >.<

dibdot, 2021-02-16:
Interesting/confusing short-hand notation ... I've found the relevant commit here: https://git.openwrt.org/?p=project/firewall3.git;a=commit;h=bba31cce0521e014109fc805671d4cff7ee9dbf6

dengqf6, 2021-02-25:
It's `iptables` that does not support this format.
```
ip6tables v1.8.7 (legacy): invalid mask `-64' specified
```
But you can just use `::ffff:ffff:ffff:ffff`

SoniEx2, 2021-02-25:
but luci is a frontend to fw3, which handles this format.

not only that but editing the raw HTML's hidden fields to force it to send the -64 works fine.

commit f64b3d50 (2021-03-01): luci-app-firewall: allow negative prefix lengths

Fixes: #4812
commit 49355255 (2021-03-01): luci-app-firewall: allow negative prefix lengths

Fixes: #4812

(cherry picked from commit f64b3d509446cc78a7df746b174bdab330f23ab2)
commit c3db6905 (2021-03-01): luci-app-firewall: allow negative prefix lengths

Fixes: #4812

(cherry picked from commit f64b3d509446cc78a7df746b174bdab330f23ab2)
closed 2021-03-01 by jow-