openwrt/luci #4813: lua to javascript migration questions

stangri, 2021-02-15:
I'll be working on migrating luci apps for https-dns-proxy, simple-adblock, vpn-policy-routing and vpnbypass. 

Just for reference I'll copy the link from a different migration issue: http://openwrt.github.io/luci/jsapi/index.html

I use the service enable/disable and start/stop/special action buttons for all my services (example: https://github.com/openwrt/luci/blob/master/applications/luci-app-https-dns-proxy/luasrc/view/https-dns-proxy/buttons.htm). My plan is to obtain the service enabled/disabled and running/stop status thru a luci rpcd script.

Questions:
* is the dependency on luci rpcd script (to obtain principal package status) necessary in this case or is there something esle I can do?
* is there anything openwrt-specific to include this buttons.js file from /htdocs/.../package-name.js?
* where do I place the actual buttons.js file, as well as required [css](https://github.com/openwrt/luci/blob/master/applications/luci-app-https-dns-proxy/luasrc/view/https-dns-proxy/css.htm) and [js](https://github.com/openwrt/luci/blob/master/applications/luci-app-https-dns-proxy/luasrc/view/https-dns-proxy/js.htm) files?
* where do I place the [custom status](https://github.com/openwrt/luci/blob/master/applications/luci-app-https-dns-proxy/luasrc/view/https-dns-proxy/status.htm) and [custom textarea status](https://github.com/openwrt/luci/blob/master/applications/luci-app-https-dns-proxy/luasrc/view/https-dns-proxy/status-textarea.htm) formatting files?
* the calls from the buttons are currently processed in the controller: https://github.com/openwrt/luci/blob/master/applications/luci-app-https-dns-proxy/luasrc/controller/https-dns-proxy.lua#L9-L27 what would be the correct way of migrating this? 

Generic migration questions:
* both in [model/cbi](https://github.com/openwrt/luci/blob/master/applications/luci-app-vpnbypass/luasrc/model/cbi/vpnbypass.lua#L91-L93) files and [controllers](https://github.com/openwrt/luci/blob/master/applications/luci-app-https-dns-proxy/luasrc/controller/https-dns-proxy.lua#L9-L27) I have calls to enable/disable and start/stop/restart services (both for my packages and sometimes dnsmasq/others) -- is there a way to initiate those from javascript or should I do that from the luci rpcd script?
* follow-up to the question above, how do I implement something to work on `on_after_commit` with javascript? I've searched within repo but only found examples inside the lua files.
* in https-dns-proxy I have a directory with files for supported providers, in Lua it was easy to have a [resource in those translatable](https://github.com/stangri/source.openwrt.melmac.net/blob/master/luci-app-https-dns-proxy/luasrc/https-dns-proxy/providers/ca.cira.canadianshield.family.lua#L3), how do I achieve the same with the [json file](https://github.com/stangri/source.openwrt.melmac.net/blob/master/luci-app-https-dns-proxy-js/root/usr/share/https-dns-proxy/providers/ca.cira.canadianshield.family.json#L3)?
* where ss the description and what else is available for tools.widgets (as seen in [luci-app-adblock](https://github.com/openwrt/luci/blob/master/applications/luci-app-adblock/htdocs/luci-static/resources/view/adblock/overview.js#L8)):

```
'require tools.widgets as widgets';
```

As suggested by @apacar I'm tagging @PolynomialDivision and @JonnyTischbein .

Ansuel, 2021-02-15:
That's a lot of question... The short answer is that you should i any way possible to use the normal luci api that are already implemented... the fs api is very useful to access system...

Also notice that you need to implement the correct acl or the package won't work...

All the hard work should be done in the rpcd api... And AFAIK create a rpcd api is the only way since it will be usable by the luci apps with the ubus interface...
Soooo if your app use massively use all the legacy stuff, a big rework is needed to adapt and simplify all using the new js api.

dibdot, 2021-02-16:
Regarding widgets look at the exposed functions in widgets.js ... I'm not aware of further docs.
https://github.com/openwrt/luci/blob/5c5b1340e5541d6283b825f7756812fc1784756f/modules/luci-base/htdocs/luci-static/resources/tools/widgets.js#L618-L624

jow-, 2021-02-19:
> I use the service enable/disable and start/stop/special action buttons for all my services (example: [...]

There's ubus APIs available for that.

To query enabled/disabled, use the `luci / getInitList` procedure, passing the init script as `name` parameter.
CLI:
```bash
root@OpenWrt:~# ubus call luci getInitList '{ "name": "dropbear" }'`
{
	"dropbear": {
		"index": 19,
		"enabled": true
	}
}
```

JS: 
```javascript

var callInitList = rpc.declare({
  object: 'luci',
  method: 'getInitList',
  params: [ 'name' ]
});

callInitList('dropbear').then(function(reply) {
  console.debug('Dropbear is ' + reply["dropbear"].enabled);
});
```

To execute init actions, use the `luci / setInitAction` procedure.

CLI:
```bash
root@OpenWrt:~# pidof ntpd
30535
root@OpenWrt:~# ubus call luci setInitAction '{ "name": "sysntpd", "action": "stop" }'
{
	"result": true
}
root@OpenWrt:~# pidof ntpd
root@OpenWrt:~# ubus call luci setInitAction '{ "name": "sysntpd", "action": "start" }'
{
	"result": true
}
root@OpenWrt:~# pidof ntpd
30816
root@OpenWrt:~# 
```

JS: 
```javascript
var callInitAction = rpc.declare({
  object: 'luci',
  method: 'setInitAction',
  params: [ 'name', 'action' ],
  expect: { result: false }
});

callInitAction('sysntpd', 'stop').then(function(reply) {
  console.debug('Stop status: ' + reply.result);
});
```

> Questions:
>     * is the dependency on luci rpcd script (to obtain principal package status) necessary in this case or is there something esle I can do?

If with "luci rpcd script" you mean a custom script in `/usr/libexec/rpcd/...` then no, it is not strictly necessary. It usually is only required if you need to expose information or actions via ubus which are not available through other methods already.

>     * is there anything openwrt-specific to include this buttons.js file from /htdocs/.../package-name.js?
[... 64 lines cut ...]
```

> Generic migration questions:
>
>     * both in [model/cbi](https://github.com/openwrt/luci/blob/master/applications/luci-app-vpnbypass/luasrc/model/cbi/vpnbypass.lua#L91-L93) files and [...]

Yes, see first reply.

>     * follow-up to the question above, how do I implement something to work on `on_after_commit` with javascript? I've searched within repo but only found examples inside the lua files.

Depends on what you want to do exactly. Right now, a `uci-applied` event is broadcasted to the document whenever a uci apply operation completed, so you could do something like

```javascript
document.addEventListener('uci-applied', function(ev) {
  alert('A uci apply operation just finished');
});
```

> * in https-dns-proxy I have a directory with files for supported providers, in Lua it was easy to have a [resource in those [...]

That might require changes to the translation infrastructure. For menu JSON files I added a special case to treat the "title" field as translation source. The same can be done for your packages, or even better we add a new file extension, like `.json.intl` which is then automatically preprocessed.

>     * where is the description and what else is available for tools.widgets (as seen in [luci-app-adblock](https://github.com/openwrt/luci/blob/master/applications/luci-app-adblock/htdocs/luci-static/resources/view/adblock/overview.js#L8)):

The `tools.widgets` class implements a few special widgets that require extra logic. The base widgets are documented in http://openwrt.github.io/luci/jsapi/LuCI.ui.html and their CBI wrappers in http://openwrt.github.io/luci/jsapi/LuCI.form.html

stangri, 2021-02-22:
Thanks @jow- this is eye-opening.

I've decided to start with the easiest WebUI -- luci-app-vpnbypass and here's the difficulties I'm facing:
* I used to have two maps (one for vpnbypass and one for dhcp, so dnsmasq ipset entries could be inserted directly into dhcp config) in [model/cbi](https://github.com/stangri/source.openwrt.melmac.net/blob/master/luci-app-vpnbypass/luasrc/model/cbi/vpnbypass.lua), I'm struggling to achieve the same [with js](https://github.com/stangri/source.openwrt.melmac.net/blob/master/luci-app-vpnbypass-js/htdocs/luci-static/resources/view/vpnbypass/overview.js#L48-L53). Can I have (and return) two maps from the single js file or am I limited to a single map and would need some trickery to detect save and write ipsets to `/etc/config/dhcp` instead of my own config file?
* In model/cbi I used to [restart dnsmasq if there were any updates to `dhcp` config](https://github.com/stangri/source.openwrt.melmac.net/blob/master/luci-app-vpnbypass/luasrc/model/cbi/vpnbypass.lua#L85-L93), do I need something like that again in js view or should dnsmasq automatically restart upon detecting changes in the `/etc/config/dhcp`?
* Also, is there a translatef equivalent in js? I want to be able to make a more complex translateable string and I probably missed it in the other js-based WebUIs I've checked.
* Because enabled/disabled depends in the uci config value and running/stopped cannot be determined by the PROCD instance I've had to create [my own custom script in /usr/libexec/rpcd/](https://github.com/stangri/source.openwrt.melmac.net/blob/master/luci-app-vpnbypass-js/root/usr/libexec/rpcd/luci.vpnbypass) (that was actually the easiest bit as I'm more familiar with shell script with anything else right now). I've tried to mimic the `getInitList` and `setInitAction` methods of luci rpcd script, and at the moment you can pass any init file name to it. Can I limit the list of parameters in ACL file and/or should I ignore the name of the script passed to luci.vpnbypass and explicitly use vpnbypass init in the RPCD?
* Follow up -- I'm planning to add another method for custom RPCD script -- getEvironment, is there a standard luci RPCD method like that I should also mimic?
* Finally, I've tried to [create my own widget](https://github.com/stangri/source.openwrt.melmac.net/blob/master/luci-app-vpnbypass-js/htdocs/luci-static/resources/vpnbypass/buttons.js) but I failed. Is there a simple example I could follow? I'll need to implement the buttons and the service status widgets re-creating functionality from mobule/cbi as below:
[image]

I'm sorry if any of the questions above are silly, I don't really know js and it's very overwhelming trying to adopt js and luci specifics. I certainly need some hand-holding to get me started and I believe I'll be able to convert my other packages then.

stangri, 2021-04-13:
Adding my own PR for client-side rendering transition for luci-app-vpnbypass: https://github.com/openwrt/luci/pull/4895.

stangri, 2021-10-29:
Super-helpful CBI lua to js script by @jow- https://github.com/openwrt/luci/issues/5461#issuecomment-951979776

stangri, 2023-09-25:
With the wealth of information available presently, there's no longer need for this, thanks to everyone who replied!

closed 2023-09-25 by stangri