openwrt/luci #4816: Bug : Viewing realtime connection causes LuCI to crash with a lot of active connections.

hagaygo, 2021-02-17:
Described on https://forum.openwrt.org/t/luci-crash-on-19-07-2-when-visiting-admin-status-realtime-connections/60674/8

Reposting my post from there 

When active connection is high (Above 2-3k) the following command

`ubus call luci getRealtimeStats '{ "mode": "conntrack" }'`

Works just fine and returns a reply.

And this

`ubus call luci getConntrackList`

Just hangs an causes Luci to return

`/usr/lib/lua/luci/dispatcher.lua:426: /etc/config/luci seems to be corrupt, unable to find section 'main'`

Only restarting rpcd gets it back using

`service rpcd restart`

My device has plenty of ram (RPI) so i think the code for returning the connection list doesnt really work with large numbers, I guess it should only return the top 100 connections or something like that.

Using my mobile app on my router (https://github.com/hagaygo/OpenWRTManager) with "Active Connections" overview causes LuCI to keep crashing while there is a lot of active connections.
It uses same API as real time page.

Using 19.07.6

wulfy23, 2021-09-30:
another producer: https://forum.openwrt.org/t/rpi4-community-build/69998/1446?u=wulfy23

ptpt52, 2021-10-30:
it is likely caused by taking too long while reading /proc/net/nf_conntrack

ptpt52, 2021-10-30:
I don't know how to fix it.

hagaygo, 2021-10-31:
Its quite easy to reproduce , just open a lot of connections using a simple script and then access the page/api.

I don't think its related to the size of the file , cause it seems to crash immediately and not like after few seconds (my guess not enough memory is allocated to load/parse the file).

Hope this helps.

ptpt52, 2021-10-31:
strace on rpcd
```
munmap(0xb5039000, 16384)               = 0
munmap(0xb5045000, 32768)               = 0
munmap(0xb5041000, 16384)               = 0
munmap(0xb502d000, 16384)               = 0
munmap(0xb5025000, 16384)               = 0
munmap(0xb5019000, 16384)               = 0
munmap(0xb5031000, 32768)               = 0
munmap(0xb5015000, 16384)               = 0
munmap(0xb5029000, 16384)               = 0
munmap(0xb501d000, 32768)               = 0
munmap(0xb5009000, 16384)               = 0
munmap(0xb5001000, 16384)               = 0
munmap(0xb500d000, 32768)               = 0
munmap(0xb5005000, 16384)               = 0
munmap(0xb4ff5000, 16384)               = 0
munmap(0xb4ff9000, 32768)               = 0
munmap(0xb4fe1000, 81920)               = 0
mmap2(NULL, 4849428, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb690a000
munmap(0xb6f66000, 4096)                = 0
sendmsg(6, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\2\0\2\230/\277\216", iov_len=8}, {iov_base="\0I\376\364\3\0\0\0100r\251\222\7I\376\350\201I\376\344\0\6result\0\0\0\0"..., iov_len=4849396}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = 183040
sendmsg(6, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="rk\0\000256\0\205\0\0\24\0\7packets\0\0\0\0\0\0\1\203\0\0\22"..., iov_len=4666364}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = -1 EPIPE (Broken pipe)
--- SIGPIPE {si_signo=SIGPIPE, si_code=SI_USER, si_pid=22704, si_uid=0} ---
sendmsg(6, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0\1\0\2\230/\277\216", iov_len=8}, {iov_base="\0\0\0\24\1\0\0\10\0\0\0\0\3\0\0\0100r\251\222", iov_len=20}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = -1 EPIPE (Broken pipe)
--- SIGPIPE {si_signo=SIGPIPE, si_code=SI_USER, si_pid=22704, si_uid=0} ---
munmap(0xb4722000, 4853760)             = 0
close(9)                                = -1 EBADF (Bad file descriptor)
close(11)                               = -1 EBADF (Bad file descriptor)
munmap(0xb6f62000, 16384)               = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=71594, tv_nsec=569873677}) = 0
epoll_pwait(3, [{EPOLLIN|EPOLLERR|EPOLLHUP|EPOLLRDHUP, {u32=3068800108, u64=3068800108}}], 10, -1, NULL, 8) = 1
epoll_ctl(3, EPOLL_CTL_DEL, 6, NULL)    = 0
recvmsg(6, {msg_namelen=0}, 0)          = -1 ECONNRESET (Connection reset by peer)
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=71594, tv_nsec=574938439}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=71594, tv_nsec=575814150}) = 0

```

ptpt52, 2021-10-31:
ubus client wait and just timeout, why no reply?

ptpt52, 2021-11-01:
ping @nbd168

dr-ni, 2022-08-16:
sorry again not working anymore - only one time working :-(

rodajc, 2022-11-02:
Bug still in OpenWrt v22.03.2

rodajc, 2022-11-02:
It reproduces perfectly here because I have a Tor relay running on a Rockchip RK3328 Quad-core Cortex-A53 which means loads of connections (> 4000). So if I can help checking anything just let me know.

dr-ni, 2022-11-02:
for me it's working with v22.03.2

rodajc, 2022-11-02:
With how many connections is that?

dr-ni, 2022-11-03:
A quick test with a script from http://marianlonga.com/simulate-parallel-connections-to-a-website/ allowed me up to 1500 connections against my router from a raspberry pi 4

[image]

hagaygo, 2022-11-03:
I can confirm it seems to be fixed on my RPI4 on 22.03.2 (maybe earlier)

[image]

No crash and both GUI and my app works fine.

closed 2022-11-03 by hagaygo

rodajc, 2022-11-03:
Bug still in OpenWrt v22.03.2

LuCI menu Status / Overview:

[image]

but LuCI menu Status / Realtime Graphs / Connections stays empty:

[image]

And after that accessing any menu in LuCI:
[image]

So rpcd crashes accessing Status / Realtime Graphs / Connections menu in LuCI with > 4000 connections approx.

hnyman, 2022-11-03:
@jow- 
Should the connection list be changed to be manually refreshed?
I am not sure if it is wise to refresh really long lists automatically every 3 seconds if the tab is open.

How about drawing it once and then offering a refresh button?

jow-, 2022-11-03:
This will not solve the issue.

jivanpal, 2024-01-25:
I seem to be experiencing this in the same way as @rodajc on OpenWrt 23.05.2 on a Netgear DGN3500. However, I never see the error message `/etc/config/luci seems to be corrupt`. Instead, LuCI is completely unresponsive until rpcd restarts.

On this device, which has 64MB RAM, by default has max connections of 7168, and in my environment usually/often has active connections in the 3000–5000 range, this issue seems to occur consistently when the connection count exceeds around 1000.

I second @hnyman's desire for a manually refreshed list of connections (I almost always just want to see the graph actively refreshing/updating, not the list of connections) and/or a capped/paginated list of connections (e.g. top 100 connections by sort order, or 100 connections per page).

garyfisher, 2024-04-07:
OpenWrt 23.05-SNAPSHOT r23821-77b7c10884 / LuCI openwrt-23.05 branch git-24.086.45142-09d5a38

Model: Xiaomi redmi AX6000

Active connection: 13515

when entering /cgi-bin/luci/admin/status/realtime/connection, page freezes and then message "session has expired" appears and it happens every time.

hyfink, 2025-08-19:
haven't fix yet on 24.10 x86