openwrt/luci #4825: luci-mod-network: the correct Wi-Fi status is not displayed

castillofrancodamian, 2021-02-20:
In a clean installation of OpenWrt without configuration files, pressing the WPS button, which activates the wireless radios, does not correctly display their status in LuCI. After a device reboot, it shows the status correctly.

[image]
[image]

castillofrancodamian, 2021-04-30:
Someone?

Ansuel, 2021-09-18:
@castillofrancodamian can you better describe the problem? more details on how to reproduce

castillofrancodamian, 2021-09-18:
I can no longer reproduce it. At that time, I have tried multiple builds and all were showing the same error.

castillofrancodamian, 2021-09-18:
They will have already fixed it, so I close this.

closed 2021-09-18 by castillofrancodamian