openwrt/luci #4826: luci-mod-network: IPv6 on wireless wan not created

plntyk, 2021-02-21:
Scenario:
<Provider> -LAN- <OpenWrt 1 w. IPv6 ; AP> -Wifi- < Client - OpenWrt 2> -LAN-

Luci -> Wireless -> Scan -> Join Network

does not add IPv6 connectivity to the new WWAN device.
Clients attached to OpenWrt 2 do not have IPv6 connectivity

additional steps to get ipv6 connectivity:
- create wwan6 interface 
- select ipv6
- select physical device wwan 
- add wwan6 to firewall zone wan/wan6/wwan

Losing ipv6 connectivity when chaining devices is unexpected - chaining ipv4 "works" (even if it is double-nat)

prateekrastogi, 2021-02-21:
Looking forward to forwarding all IPv4 Requests to IPv6 for tunnelling Internet, i am also looking for methodologies to chain IPv6 without IPv4 or link state ppp famously called wan.

aparcar, 2022-05-30:
I think a better solution would be to create `wwan6` automatically so a user would end up with `wan`, `wan6`, `wwan` and `wwan6`. IPv6 should be enabled by default in my opinion.