openwrt/luci #4838: rpcd-mod-luci: Return array of IPv4/IPv6 addresses in getHostHints

nwidger, 2021-02-26:
The getHostHints method in rpc-mod-luci seems to go to a lot of trouble to find all IP addresses associated with hosts connected to OpenWRT, but the response only allows returning one IPv4 and one IPv6 address per host.  There doesn't seem to be any prioritization of which addresses are returned if a host has multiple IPv4 or (more likely) multiple IPv6 addresses, instead it seems to be the case that the first address found is the one returned.  It can also lead to inconsistent results, for instance if a host has both a SLAAC and DHCPv6 IPv6 address, getHostHints may return a different IPv6 address for the host depending on which address appears first in the ND table.

Was it a conscious decision to only return one address per address family?  Would it be possible to add new i.e. `ipv4_addrs` and `ipv6_addrs` array fields which return _all_ IPv4/IPv6 addresses found for a host?  If so I'd be happy to take a stab at this.

jow-, 2021-03-01:
The reasoning to return only one IP was influenced by the way the procedure is currently used. Since these hints are typically consumed by a combobox/dropdown style ui widget a one host->one IP relationship did fit. In general I do agree that it makes sense to return all IP addresses.

If we opt to go the array route, all downstream consumers should be adjusted to loop all IP address values returned. It would also be great if the returned array would be sorted in a meaningful way. I think we could assign priorities to the address objects gathered internally, e.g. give a prio of 200 to configured static leases, a prio of 100 to address records read from DHCP lease state, a prio of 50 to records from ether, 10 to IP neighbour entries etc.

nwidger, 2021-03-01:
So if I'm interpreting your response correctly, would the updated ubus response format look something along the lines of:

``` json
{
    "00:00:00:00:00:00": {
	"ipv4": "192.168.100.1",
	"ipv6": "2001::1",
	"name": "foo.lan",
	"ipv4_addrs": ["192.168.100.1", "192.168.100.2", "192.168.100.3"],
	"ipv6_addrs": ["2001::1", "2001::2", "2001::3"]
    }
}
```

where `ipv4_addrs` and `ipv6_addrs` are sorted by address priority (in descending order I would assume?).  Would we also want to update the existing `ipv4` and `ipv6` fields to contain the highest priority address for that address family?  I'm wasn't if that would break backwards-compatibility in any way.

In terms of updating downstream consumers, would all references that need updating be contained within this repo, or are there others that need to be checked?

jow-, 2021-03-01:
I think the only downstream users are in this repo (the procedure is provided by luci.so after all). For me it would also be fine to drop the "ipv4" and "ipv6" fields entirely in favor to the new array format.

nwidger, 2021-03-01:
@jow- I've opened the draft PR that just updates rpcd-mod-luci with the changes we discussed.  I was hoping to get at least a tentative "looks good" before diving into updating the downstream users of getHostHints.  I tried to assign reasonable looking priorities based on the address source, but tweaking those is easy is you think they should be in any way different.

This is my first (real) contribution so I'd appreciate any feedback. :)

nwidger, 2021-03-19:
@jow- Just wondering if you've had a chance to look at the draft PR.  In the meantime, I've taken a look at the frontend code.  I think I've wrapped my head around what would need to be changed and it doesn't look like it should be too bad.

jow-, 2021-03-19:
@nwidger - yep, apart from various white space quirks the code looked good to me. Can you reformat the file consistently with tab indentation? That aside, I think we can go forward with it.

nwidger, 2021-03-19:
@jow- Excellent, glad it overall looks good.  I noticed that issue and thought I had reindented all my changes to use tabs a few weeks ago.  I was a little sneaky and force pushed, can you look at the latest commit (dd55edae1) in the PR let me know if you still see indentation issues?

commit a5195e78 (2021-06-03): rpcd-mod-luci: Return array of addresses in getHostHints

Update luci-rpc's getHostHints method to return two string arrays for
each host, `ipaddrs` and `ip6addrs`, each containing the host's IPv4
and IPv6 addresses, respectively.  Each array is sorted by a priority
derived from the source from which the address was discovered.  The
current address sources and their priority is as follows (a
higher (larger) priority is listed first):

    - neighbor table entries: 10
    - /etc/ethers entries:    50
    - DHCP leasefile:        100
    - RRDNS queries:         100
    - getifaddrs():           [... cut]
commit 87bcf9e9 (2021-06-12): rpcd-mod-luci: Return array of addresses in getHostHints

Update luci-rpc's getHostHints method to return two string arrays for
each host, `ipaddrs` and `ip6addrs`, each containing the host's IPv4
and IPv6 addresses, respectively.  Each array is sorted by a priority
derived from the source from which the address was discovered.  The
current address sources and their priority is as follows (a
higher (larger) priority is listed first):

    - neighbor table entries: 10
    - /etc/ethers entries:    50
    - DHCP leasefile:        100
    - RRDNS queries:         100
    - getifaddrs():           [... cut]
closed 2021-06-03 by jow-