openwrt/luci #4839: RFC: Adding system description (short) and system annotations (long) fields to UCI system section

hmh, 2021-02-26:
I'd like to propose adding two new fields to UCI's system config.   I am also offering to do all the work required to implement the required changes in LuCI (for all branches) as well as updating the openwrt wiki.

This is actually filling an usability niche we detected, and which would be much better addressed in openwrt itself than in a downstream fork.

If the proposal below is acceptable, I'll submit the PRs.  Otherwise, I'd like to know what should be changed so that we can have equivalent functionality in OpenWRT upstream.  If this should be handled through the mailing lists instead, please say so.  I sent it here because it is 99,9% a LuCI / UI matter.

**system.@system[0].description = 'an optional, short description for this system, one line, some reasonable char limit'**

The "description" option is purposefully short, so that it can be used e.g. on UIs to help describe this device in a selector.  The  hostname is often not available for such purposes, or not descriptive enough, and it is better not not overload it with too much functionality.

It would use the LuCI "Value" option type, so embedded newlines would be discarded/removed.

**system.@system[0].annotations = 'an optional text for annotations in free-form,
such as install notes, location notes, serial number of the unit,
maybe inventory numbers, or telephone number of tech support'**

The "annotations" option is multi-line, and can be used to store general free-form metadata *for human consumption*.  It would use the LuCI "TextValue" option type.

These two new options are very simple and very safe to implement, as they have no side-effects on the rest of the system (well, it adds a few i18n strings, but that's it) and require no specific handling.

Besides implementing them and submitting them through a PR for LuCI (for all branches that are still accepting PRs), I would do the work to document them (should they be accepted) officially in the main wiki in pages where UCI system is described, so that one can actually extend packages to use system.@system[0].description where it make sense to do so, and to avoid duplicate work.

jow-, 2021-03-01:
Fine with me.

hmh, 2021-03-01:
Apparently, "notes" is a more appropriate word choice than "annotations" for the intended usage of the field I proposed, so I am renaming it from "annotations" to "notes".

I will submit PRs for master/trunk, 21.02, 19.07 and 18.06 -- we will also have one for 15.05, but I doubt it makes sense to submit it as a PR.

hmh, 2021-03-01:
The changes for 18.06 were also tested on 15.05, and work well enough (lua backend). Tested on 15.05 and 18.06.
If PRs for 15.05 and 17.01 are desired, just say so.

The changes for 19.07, 21.02 and master are identical (js frontend). Tested on 19.07.

hmh, 2021-03-01:
After any changes due to reviews/comments, and if the PRs are accepted, I will update the wiki pages accordingly.

nickberry17, 2021-04-01:
I like this, nice idea

stangri, 2021-04-01:
@hmh, you may want to prepend the subjects for PRs for releases with release number for clarity, like this: `[18.06] luci-mod-system: implement system.description, system.notes`.

hmh, 2021-04-01:
@stangri: done. For the LuCI development trunk (master/main/whatever) PR, I used [snapshot]...

hmh, 2021-04-01:
Updated the issue title: removed the RFC since the comments were positive, and changed the issue title to mention "notes" instead of "annotations" (I had corrected it already in the description and the PRs, but forgot about the title).

hmh, 2021-04-10:
Since the PRs were accepted, I am now updating the wiki.

The following page has been updated:
https://openwrt.org/docs/guide-user/base-system/system_configuration#system_section

I am looking for other pages that would benefit from edits re. this matter, please feel free to suggest any you see fit.

closed 2021-10-19 by feckert