openwrt/luci #4848: Host Showing IPV6 Address or ? Mark / Intermittent IPV4 Address

manjotsc, 2021-02-28:
Hi,

Setup the router as Access Point and connect devices using WiFi. Then go under Network → Wireless then look under HOST there will be a IPV6 address or a question mark.

- R7800, R6850
- OpenWrt 19.07.7 r11306-c4a6851c72, OpenWrt 21.02-SNAPSHOT r15863-3

https://forum.openwrt.org/t/host-showing-ipv6-address/89804/19

[image]
feef9c555

prateekrastogi, 2021-03-01:
Goodluck if you can disconnect link PPP of WAN, not showing under any UI using IPv4 bridge 0.0.0.0 despite all Packets being Port Forwarded to of Public IPv6 Internet Addresses dual pair.

jow-, 2021-03-01:
Not a bug. In pure AP mode, the device is not involved in layer 3 routing, hence no details about used IPs or hostnames. Wontfix.

closed 2021-03-01 by jow-