openwrt/luci #4852: Please show full absolute values instead of short form

saudiqbal, 2021-02-28:
Short values like 4K interfere in troubleshooting and analysis, please show full values like 4501 Bytes in LUCI.

jow-, 2021-03-01:
Too vague to act upon

closed 2021-03-01 by jow-

lantis1008, 2021-03-01:
Context: https://forum.openwrt.org/t/is-it-possible-to-see-absolute-values-in-openwrt/87291
I think OP wants packet counters in the firewall sections of LUCI to show full values (not order of magnitude abbreviations).

saudiqbal, 2021-03-01:
> Too vague to act upon

> Context: https://forum.openwrt.org/t/is-it-possible-to-see-absolute-values-in-openwrt/87291
> I think OP wants packet counters in the firewall sections of LUCI to show full values (not order of magnitude abbreviations).

That's correct, please reopen this issue.
Thanks.

commit 52caa1dc (2021-03-01): luci-mod-status: allow displaying raw iptables counter values

Fixes: #4852
commit b39b7fc0 (2021-03-01): luci-mod-status: allow displaying raw iptables counter values

Fixes: #4852