openwrt/luci #4865: Overview page does not work on Safari (macOS)

oskarirauta, 2021-03-02:
Overview page has indefinitely spinning spinner on Safari for OS X.
XHR's are not loading either.

Works on Chrome (for macOS) but not on Safari.

oskarirauta, 2021-03-03:
Issue is with 30_network.js and most likely on it's minifying; I downloaded it's raw version and did not minify it and overview page started working again.

jow-, 2021-03-03:
I don't have access to any OS X so can't really test. Any hint in the browser debug console?

oskarirauta, 2021-03-03:
File loading stalled in middle of it. No debug results as file loading did not complete. After a long time, luci reports that browser gave up on request.
 Version without minifying works.

Lähetetty iPhonesta

jow-, 2021-03-03:
Do you use `luci-ssl` / HTTPS? Does the same happen without HTTPS?

oskarirauta, 2021-03-03:
Did not try with https. Only plain http was tested.

closed 2021-11-29 by oskarirauta